realsenseai/librealsense #5445: T265: A question regarding relocalization

neilyoung, 2019-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

I noticed, that it sometimes goes very quickly but _if_ there is a relocation, it always happens within 30 seconds. Is this "30 seconds" window something intrinsic to the algorithm?

Even if the area is well "trained", it can take up to 30s until the relocation happens.

MartyG-RealSense, 2019-12-13:
There is an option called **RS2_OPTION_ENABLE_RELOCALIZATION** which I believe is True by default.  The T265 documentation says about it:

"This allows the device to solve the Kidnapped Robot Problem, i.e. it will allow connecting the current map to a loaded map or connecting the current map to itself after accumulating a large drift, say after closing a large loop or covering the camera and walking around.  This is independent of jumping the pose.  When fooled this feature can lead to much larger errors than are likely via the basic mapping".

I wonder if the speed at which your robot is travelling causes drift to accumulate at such a pace that it accumulates enough drift to always trigger a relocalization before or on the 30 second point from the time that travel begins.

neilyoung, 2019-12-13:
@MartyG-RealSense The robot is me :)

I was just wondering if there would be a process triggered each 30 secs, which does the relocation

MartyG-RealSense, 2019-12-13:
My understanding is that drift is not necessarily tied to continuous robot motion and that it can also accumulate when manually moving the camera from the start position (something that someone might do when running accuracy tests).  

I am not a T265 specialist.  As far as I know though, resets are primarily caused in response to real-time conditions and events rather than a formal timer (covering up the camera, travelling in a circle, accumulating excessive drift, etc).  Apparently, according to the Visual SLAM documentation, loss of power may be another trigger condition for relocalization.

Are you moving into another room when travelling or staying in the same room?  In SLAM systems, travelling to other rooms can cause drift to accumulate until the robot returns to the start point and relocalizes.

neilyoung, 2019-12-13:
What I'm noticing is, that the relocation takes place very quickly, if I resume from the initial start point of the map. If I'm moving to somewhere in the field, into an area, which is already mapped, then the relocation happens later, but not later than 30 s. It might be a subjectively erroneous monitoring.

MartyG-RealSense, 2019-12-13:
A T265 case-study document by Intel describes relocalization as follows:

"Re-localization is the ability of the camera to use the features it has seen before to recognize when it has returned to a familiar place.  For example, if a drone makes an aerial loop and then returns to its original position, the algorithm is able to compare the visual environment against its database of salient features and their positions in space to re-localize.  The camera is able to discern its position against that static frame of reference, locating its point of origin with an error margin of less than one percent".

neilyoung, 2019-12-13:
Yes, sure. I'm aware about what relocation is. I was just trying to recognize a pattern in the observed behavior. Since it is not documented elsewhere it would be very helpful to have a statement regarding how long it will take at maximum, if there is a realistic chance of a relocation (means: The area has been carefully mapped before and the map is used).  Look, say you are doing this mapping really carefully, store the map and provide that to other robots, there should be at least a minimal estimate about how long it could take at maximum under ideal conditions. I'm also aware, that lighting, changing environments and so on can make this relocation harder.

MartyG-RealSense, 2019-12-13:
I do appreciate your frustration.  I do not have a T265 myself to perform tests with, so my accumulation of T265 knowledge is based on learning a bit more with each T265 support case I take part in.  Sometimes, the best outcome I can provide is to draw out information that the next person who provides support can use in their diagnostic process.    Your latest questions are outside of my T265 knowledge unfortunately, so hopefully one of the Intel folk on this forum will be able to use the information discussed so far to assist you.  I do apologize and thank you for your patience!

neilyoung, 2019-12-13:
@MartyG-RealSense I'm absolutely not frustrated :) I very much appreciate your patience and responsiveness and what you are doing in this community. Sorry, if it appeared so as if I would be frustrated. This is not the case :)

neilyoung, 2019-12-13:
Closing not on purpose

dorodnic, 2019-12-16:
From my (also limited) knowledge of T265 inner workings, IMHO there is no special significance to the 30sec time window. I have vague memory of @radfordi mentioning that relocalization loop got optimized from 1200 ms to 600 ms in #5234, but I may be way off on this. Jim, could you please comment?

radfordi, 2019-12-16:
The device tries to relocalize a few times a second, so any perceived 30s number has no special significance.

neilyoung, 2019-12-17:
@radfordi Thanks for clarification. Just to not construct things, which are not there :)

closed 2019-12-17 by neilyoung