realsenseai/librealsense #5447: T265: Speed up pose stream playback from bag file

neilyoung, 2019-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

I'm able to playback a ROSbag file and derive poses from that. Since I know, that I can speed up the playback with realsense-viewer, I was trying to do the same in my python script to no avail. Only 200 is accepted for the frame rate parameter. Is there any API I could use to speed up? Especially for long recordings it is a bit cumbersome to wait in real-time for the processing.

Only this works:

`config.enable_stream(rs.stream.pose, rs.format.six_dof, 200)
`
Setting it to 400 or so excepts with `Couldn't resolve requests`.

thien94, 2019-12-15:
I believe for rosbag you can set the `rate` parameter to speed up the playback speed, see [here](https://wiki.ros.org/rosbag/Commandline#play). Is this what you are looking for?

neilyoung, 2019-12-15:
Thanks @thien94  but no, that’s not what I was looking for. I’m playing back with the a.m.  code line from a Python script using the RealSense SDK. This line is the only contact I‘m having with the bag

bfulkers-i, 2019-12-16:
@neilyoung I believe you want `set_real_time` like:

```
import pyrealsense2 as rs

pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file("poses.bag", repeat_playback = False)
cfg.enable_stream(rs.stream.pose)
pipe.start(cfg)

profiles = pipe.get_active_profile()
dev = profiles.get_device()

playback = dev.as_playback()
playback.set_real_time(False)

try:
    while(True):
        # Wait for the next set of frames from the camera
        (success, frames) = pipe.try_wait_for_frames(1000)
        if not success:
            print("Finished")
            break

        pose = frames.get_pose_frame()
        if pose:
            # Print some of the pose data to the terminal
            data = pose.get_pose_data()
            print("Position: {}".format(data.translation))

finally:
    pipe.stop()
```

@dorodnic may have a better way to handle faster than realtime playback without repeating, but this is probably a good place to start

neilyoung, 2019-12-16:
@bfulkers-i Thanks, that looks good to me. set_real_time(False) mens full speed? Where is all that documented?

dorodnic, 2019-12-16:
Documentation for the python module is available [here](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html#module-pyrealsense2)

closed 2019-12-17 by neilyoung