realsenseai/librealsense #5449: The SDK download speed is always zero or the download fails

wuhaoqi-hub, 2019-12-14:
I can't download https://git hub.com/intelrealsense/librealsense/releases/download/v2.30.0/intel.realsense.sdk-win7-2.30.0.1174.exe on git hub successfully. The download speed is always zero or the download fails. Can you send a copy of the file to my email? My email is 742842498@qq.com, thank you！

dorodnic, 2019-12-16:
Hi @wuhaoqi-hub 
Sorry, emailing the SDK installer is very impractical given it is ~500mb. I hope you manage to figure out what is wrong with the network connection, but there is also the alternative of downloading the (much smaller) [source code](https://github.com/IntelRealSense/librealsense/archive/master.zip) and building the SDK from source.

wuhaoqi-hub, 2019-12-18:
I've solved the problem using Free Download Manager 5，Thank you for your reply.There may be some questions about the use of the  D435 camera in the future. I hope you can help me when it comes,thank you!

closed 2019-12-18 by wuhaoqi-hub