realsenseai/librealsense #5450: Fail to set D435i to High Accuracy

shadowdk3, 2019-12-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

Required Info |  
-- | --
Camera Model | D435i
Firmware Version | 5.11.11
Operating System & Version | Ubuntu16.04
Kernel Version (Linux Only) | 4.15.0-42-generic
Platform | PC
SDK Version | 2.30.0
Language | python
Segment | Robot

### Issue Description
How to set visul preset to high accuracy?
I can get visul preset range and value description, but fail to set.
`depth_sensor.set_option(rs.option.visual_preset, 3)`
and return runtime error.
`RuntimeError: set_pu(id=4) failed! Last Error: Input/output error`

MartyG-RealSense, 2019-12-14:
The link below has a Python script for setting the High Accuracy preset.

https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634

shadowdk3, 2019-12-16:
> The link below has a Python script for setting the High Accuracy preset.
> 
> [#2577 (comment)](https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634)

Yes, I also try that code, but it still return runtime error: 
`RuntimeError: set_pu(id=4) failed! Last Error: Input/output error`

MartyG-RealSense, 2019-12-16:
It sounds as though it might be related to a previously reported problem with changing presets that gave a similar runtime error with the same ID number.  The problem seemed to be particular to Linux.

https://github.com/IntelRealSense/realsense-ros/issues/390

Hopefully a member of the Intel RealSense team on this forum will be able to provide further guidance

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

Hi.

Are you still experiencing this? Just tried this setting and appears to work on latest firmware/sdk. FYI if we don’t hear from you in 7 days, this issue will be closed.

Thank you

closed 2020-04-29 by RealSenseCustomerSupport