realsenseai/librealsense #5451: ROS Message Extrinciscs

scottnortman, 2019-12-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.01 | 
| Operating System & Version |  Ubuntu 16.04 LTS + ROS Kinetic | 
| Kernel Version (Linux Only)    |  4.15.0-72-generic                                       | 
| Platform                                 |  PC (Dell Inspiron 2330)  |
| SDK Version                            |  Latest GIT pull as of 2019-DEC-13                    | 
| Language                            |  ROS Kinetic + Python                          | 
| Segment			|  Robotics                         | 

### Issue Description

I am able to launch the ROS Node using:

$ roslaunch realsense2_camera rs_rgbd.launch

In a separate terminal window, I am able to get a list of streaming topics as expected
$ rostopic list

However, when I examine the RO topics for the extrinsics of the infrared depth cameras, both 1 and 2,  the rotation matrix is identity and the translation is 0 for both infrared cameras.  It seems they are not reporting the correct extrinsics values:

$ rostopic echo /camera/extrinsics/depth_to_infra1 
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_infra1_extrinsics"
rotation: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
translation: [0.0, 0.0, 0.0]
---

$ rostopic echo /camera/extrinsics/depth_to_infra2
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_infra2_extrinsics"
rotation: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
translation: [0.0, 0.0, 0.0]
---
Why are these not populated with the correct values?

dorodnic, 2019-12-16:
@doronhi - please take a look, `rostopic echo /camera/extrinsics/depth_to_infra2` translation should not be zero. 
@scottnortman - do you get the right values using SDK tools such as `rs-enumerate-devices -c`?

scottnortman, 2019-12-16:
Thanks for your prompt response.  Here is the output from the above command:

**```

Extrinsic Parameters:
Extrinsic from "Depth"	  To	  "Depth" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Depth"	  To	  "Color" :
 Rotation Matrix:
   0.999961        -0.00325558       0.00817256    
   0.00324252       0.999993         0.00161064    
  -0.00817775      -0.00158408       0.999965      

 Translation Vector: 0.015096090734005  -0.00012191944551887  -0.000253041507676244  

Extrinsic from "Depth"	  To	  "Infrared" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Depth"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Depth"	  To	  "Infrared 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Color"	  To	  "Depth" :
 Rotation Matrix:
   0.999961         0.00324252      -0.00817775    
  -0.00325558       0.999993        -0.00158408    
   0.00817256       0.00161064       0.999965      

 Translation Vector: -0.0150971803814173  0.000170664279721677  0.000129855354316533  

Extrinsic from "Color"	  To	  "Color" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Color"	  To	  "Infrared" :
 Rotation Matrix:
   0.999961         0.00324252      -0.00817775    
  -0.00325558       0.999993        -0.00158408    
   0.00817256       0.00161064       0.999965      

 Translation Vector: -0.0150971803814173  0.000170664279721677  0.000129855354316533  

Extrinsic from "Color"	  To	  "Infrared 1" :
 Rotation Matrix:
   0.999961         0.00324252      -0.00817775    
  -0.00325558       0.999993        -0.00158408    
   0.00817256       0.00161064       0.999965      

 Translation Vector: -0.0150971803814173  0.000170664279721677  0.000129855354316533  

Extrinsic from "Color"	  To	  "Infrared 2" :
 Rotation Matrix:
   0.999961         0.00324252      -0.00817775    
  -0.00325558       0.999993        -0.00158408    
   0.00817256       0.00161064       0.999965      

 Translation Vector: -0.0150971803814173  0.000170664279721677  0.000129855354316533  

Extrinsic from "Infrared"	  To	  "Depth" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared"	  To	  "Color" :
 Rotation Matrix:
   0.999961        -0.00325558       0.00817256    
   0.00324252       0.999993         0.00161064    
  -0.00817775      -0.00158408       0.999965      

 Translation Vector: 0.015096090734005  -0.00012191944551887  -0.000253041507676244  

Extrinsic from "Infrared"	  To	  "Infrared" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared"	  To	  "Infrared 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: -0.0550608672201633  0  0  

Extrinsic from "Infrared 1"	  To	  "Depth" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared 1"	  To	  "Color" :
 Rotation Matrix:
   0.999961        -0.00325558       0.00817256    
   0.00324252       0.999993         0.00161064    
  -0.00817775      -0.00158408       0.999965      

 Translation Vector: 0.015096090734005  -0.00012191944551887  -0.000253041507676244  

Extrinsic from "Infrared 1"	  To	  "Infrared" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared 1"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared 1"	  To	  "Infrared 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: -0.0550608672201633  0  0  

Extrinsic from "Infrared 2"	  To	  "Depth" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0.0550608672201633  0  0  

Extrinsic from "Infrared 2"	  To	  "Color" :
 Rotation Matrix:
   0.999961        -0.00325558       0.00817256    
   0.00324252       0.999993         0.00161064    
  -0.00817775      -0.00158408       0.999965      

 Translation Vector: 0.0701548308134079  5.66163580515422e-05  -0.000703315658029169  

Extrinsic from "Infrared 2"	  To	  "Infrared" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0.0550608672201633  0  0  

Extrinsic from "Infrared 2"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0.0550608672201633  0  0  

Extrinsic from "Infrared 2"	  To	  "Infrared 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  
```**

This

scottnortman, 2019-12-16:
[rs_output_cal.txt](https://github.com/IntelRealSense/librealsense/files/3970612/rs_output_cal.txt)

I've attached the entire output as a text file

scottnortman, 2019-12-16:
Did not mean to close issue

dorodnic, 2019-12-17:
Thanks @scottnortman we will look into it

kevindehecker, 2019-12-17:
I'm also having this problem, I'm reasonably sure it is related to the 2.31 release. I can reproduce this by doing the following:
- with the realsense viewer, make a small recording with only both IR streams on
- then, if I load the bag file and check the extrinsics with:
```
    rs2_extrinsics e = infared1.get_extrinsics_to(infared2);
    float baseline = e.translation[0];
```
That gives me a 0.
- if I run above code on the hardware realsense instead of the bag file, the baseline says 0.0499...

scottnortman, 2019-12-23:
Hi @dorodnic , any updates on this issue?

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

Hi scottnortman,

Please check if this PR https://github.com/IntelRealSense/librealsense/pull/5521 has fixed the issue and let us know if you need any further assistance with this matter.

Thank you!

closed 2020-01-30 by RealSenseCustomerSupport