realsenseai/librealsense #5452: What is underlying USB data transfer format for left imager when set to Y8?

kafan1986, 2019-12-15:
Required Info |  
-- | --
Camera Model | { D415}
Firmware Version | {05.11.15.00}
Operating System & Version | {Android}
Platform | {Arm-Odroid}
SDK Version | { 2.30.0 }
Language | {C++/Java }
Segment | {others }

Given the android sdk stability currently is low due to high CPU usage by RSUSB data transfer and lower data transfer improves stability to an extent. I was thinking about using gray scale image (Y8) from the infrared camera.

Given the D415 left imager can support both Y8 and RGB8 format. Can someone tell me what is the native format for data transfer for infrared cam stream? Or maybe it supports both not sure though. If it is Y8 then is makes sense for me, as it will halve the data transfer compared to colour camera native format of YUYV, if not then the data transfer size remains same as colour camera.

dorodnic, 2019-12-16:
Hi @kafan1986 
Y8 format should be mapped directly to `GRAY` fourcc code, encoding 1-bpp grayscale image.

kafan1986, 2019-12-16:
> Hi @kafan1986
> Y8 format should be mapped directly to `GRAY` fourcc code, encoding 1-bpp grayscale image.

That means the data transfer requirement over USB is half of that colour image (YUYV). Am I right?

dorodnic, 2019-12-16:
Right

closed 2019-12-16 by kafan1986