realsenseai/librealsense #5453: How to set Option params for depth sensor Java (Android)?

vshalkov, 2019-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.2                                         | 
| Platform                                 | Android  |
| SDK Version                            |  RealSense SDK for Android OS                          | 
| Language                            |  C, opencv, kotlin, java                          | 
| Segment			|  Smartphone                          | 

### Issue Description
I have that code c++ :
```
// search for the depth sensor
	int depth_sensor_index_ = -1;

	for(size_t i = 0; i < sensors.size(); ++i)
		if(sensors[i].as<rs2::depth_sensor>())
		{
			MAssert(depth_sensor_index_ < 0, "there must be only one depth sensor");

			depth_sensor_index_ = i;
		}

	MAssert(depth_sensor_index_ >= 0, );

	rs2::depth_sensor depth_sensor = sensors[depth_sensor_index_].as<rs2::depth_sensor>();

	MAssert(depth_sensor, );

	_depth_scale = depth_sensor.get_depth_scale() * 1000.f;
	MAssert(_depth_scale > 0, );

	std::cout << "depth_scale: " << _depth_scale << " millimeters" << std::endl;

	// set depth sensor parameters
	{
		const std::pair<rs2_option, float> set_options[] = {
			{ RS2_OPTION_LASER_POWER,          16 },
			{ RS2_OPTION_ACCURACY,              3 },
			{ RS2_OPTION_MOTION_RANGE,         30 },
			{ RS2_OPTION_FILTER_OPTION,         5 },
			{ RS2_OPTION_CONFIDENCE_THRESHOLD, 15 },
		};

		const int set_cnt = sizeof(set_options) / sizeof(set_options[0]);

		for(int i = 0; i < set_cnt; ++i)
		{
			const rs2_option option = set_options[i].first;
			const float value = set_options[i].second;

			// MAssert(depth_sensor.supports(option), );
			if (!depth_sensor.supports(option))
			{
				std::clog << "option " << i << " is not supported\n";
				continue;
			}

			MAssert(!depth_sensor.is_option_read_only(option), );

			depth_sensor.set_option(option, value);

			MAssert(depth_sensor.get_option(option) == value, );
		}
	}
```
I want to sent this params for depth sepnsor:
```
{ RS2_OPTION_LASER_POWER,          16 },
			{ RS2_OPTION_ACCURACY,              3 },
			{ RS2_OPTION_MOTION_RANGE,         30 },
			{ RS2_OPTION_FILTER_OPTION,         5 },
			{ RS2_OPTION_CONFIDENCE_THRESHOLD, 15 },
```
If i use this code:
```
val aling = Align(StreamType.COLOR)
aling.setValue(Option.LASER_POWER, 16f)
aling.setValue(Option.ACCURACY, 2f)
aling.setValue(Option.FILTER_OPTION, 5f)
aling.setValue(Option.CONFIDENCE_THRESHOLD, 15f)
```
It's don't work because i get this error:
```
java.lang.RuntimeException: object doesn't support option #13
```
How can I set the parameters for the depth sensor? please, help

vshalkov, 2019-12-15:
I found:
```
 Config config = new Config();
        try(DeviceList devices = mRsContext.queryDevices()) {
            if (devices.getDeviceCount() == 0) {
                return;
            }

            try (Device device = devices.createDevice(0)) {
                if(device == null){
                    Log.e(TAG, "failed to create device");
                    return;
                }

                if (!device.isInAdvancedMode()){
                    device.toggleAdvancedMode(true);
                    if (CommonConstants.getInstance().DEBUG) {
                        Log.d(TAG, "Device not in advance mode. Set to advance mode now!");
                    }
                } else {
                    if (CommonConstants.getInstance().DEBUG) {
                        Log.d(TAG, "Device already in advance mode");
                    }
                }

                try {
                    byte[] presetData = getByteArrayFromResource(FacialProcessingApp.getContext(), R.raw.high_res_high_density_preset, "high_res_high_density_preset.json");
                    device.loadPresetFromJson(presetData);
                    Log.d(TAG, "Preset Set!");
                } catch (IOException e) {
                    Log.e(TAG, "failed to set preset, failed to open preset file, error: " + e.getMessage());
                } catch (Exception e) {
                    Log.e(TAG, "failed to set preset, error: " + e.getMessage());
                }

                config.disableAllStreams();
                config.enableStream(StreamType.COLOR, 0, CommonConstants.getInstance().colourStreamWidth, CommonConstants.getInstance().colourStreamHeight, StreamFormat.YUYV, CommonConstants.getInstance().depthCamFPS);
                config.enableStream(StreamType.DEPTH,0,CommonConstants.getInstance().depthStreamWidth,CommonConstants.getInstance().depthStreamHeight,StreamFormat.Z16,CommonConstants.getInstance().depthCamFPS);
            }
```
But I don’t know what Json should look like, help shape please.
I need a preset with these parameters:
```
{ RS2_OPTION_LASER_POWER,          16 },
{ RS2_OPTION_ACCURACY,              3 },
{ RS2_OPTION_MOTION_RANGE,         30 },
{ RS2_OPTION_FILTER_OPTION,         5 },
{ RS2_OPTION_CONFIDENCE_THRESHOLD, 15 },
```

kafan1986, 2019-12-15:
> I found:
>
> ```
>  Config config = new Config();
>         try(DeviceList devices = mRsContext.queryDevices()) {
>             if (devices.getDeviceCount() == 0) {
>                 return; [...]

loadPresetFromJson is used to set advance parameters. For example high resolution high density preset etc.

For your requirement, may be something like this might work. I have not tested it.

```
                   mDeviceList = mRsContext.queryDevices();

                    if (mDeviceList.getDeviceCount() == 0) {
                        return;
                    }

                    mDevice = mDeviceList.createDevice(0);
                    if (mDevice == null) {
                        Log.e(TAG, "failed to create device");
                        return;
                    }

                    Sensor mSensor = mDevice.querySensors().get(0);
                    mSensor.setValue(Option.LASER_POWER,16f);
                    mSensor.setValue(Option.ACCURACY,2f);
                    mSensor.setValue(Option.FILTER_OPTION,5f);
                    mSensor.setValue(Option.CONFIDENCE_THRESHOLD,15f);
```

vshalkov, 2019-12-15:
@kafan1986 
Yes, this should work, but in the librealsense library for android, there is no DepthSensor class :(

[image]
Sensor class does not have setValue() method

Extension enum has:
[image]

I think I need to create the DepthSensor class myself and write JNI methods for it that will execute the setValue () methods
I think correctly?

vshalkov, 2019-12-15:
@kafan1986 I update lib and it's work, thank you!!!
But i have new error.
if i sets this params, I get errors:
```
        mSensor.setValue(Option.CONFIDENCE_THRESHOLD, 15f)
        mSensor.setValue(Option.FILTER_OPTION, 5f)
        mSensor.setValue(Option.ACCURACY, 2f)
```
```
java.lang.RuntimeException: object doesn't support option #17
```

LASER_POWER works correctly
```
mSensor.setValue(Option.LASER_POWER,16f);
```

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@vshalkov The error should be due to CONFIDENCE_THRESHOLD not supported on D435. Please remove this and try again. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@vshalkov  Did you get chance to try removing "CONFIDENCE_THRESHOLD"? Any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@vshalkov Any update from your side? If we don’t hear from you in 7 days, this issue will be closed. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@vshalkov Sorry we didn't hear from you for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have questions or issues. Thanks!

closed 2020-02-17 by RealSenseCustomerSupport