realsenseai/librealsense #5454: Infared just shows green on Kali Linux dual boot with Win 10

akinsey2468, 2019-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Kali 2019.4 AMD64  | 
| Kernel Version (Linux Only)    |  5.3.0 | 
| Platform                                 | PC/Surface Pro 2017  |
| SDK Version                            |   2.31.0                         | 
| Language                            |   Linux bash I think?     | 
| Segment			|   Just trying to use the infrared camera.   | 

### Issue Description
So the camera isn't putting out any other output except for regular RGB camera footage. The DKMS didn't install, but somehow I can get the depth and rgb footage from the SDK viewer. I know it's not a camera issue because it works with my windows 10 PC with windows hello. If anybody knows why, I'd be very grateful.

MartyG-RealSense, 2019-12-15:
There has been a previous case reported of SR300's infrared not working in programs other than the RealSense Viewer.  @dorodnic explains why in the discussion.

https://github.com/IntelRealSense/librealsense/issues/2715

The SR300 has had Windows Hello support in the past, though the newer 400 Series cameras did not.

dorodnic, 2019-12-16:
@akinsey2468 I'm confused. You say `the camera isn't putting out any other output except for regular RGB camera footage` but also `I can get the depth and rgb footage from the SDK viewer`. The title states `Infared just shows green`.

* Are you able to get depth data from the SDK / Viewer?
* What are you getting when enabling IR stream in RealSense Viewer? Perhaps you could upload screenshot from the Viewer when enabling all streams (you can drag&drop picture into GitHub comment box)

Kernel 5.3 should be sufficient to work without the DKMS.

ev-mp, 2019-12-17:
@akinsey2468 , with regards to the green output - please check whether [this fix](https://github.com/IntelRealSense/librealsense/issues/321#issuecomment-353867626) could be applicable in your case.

closed 2019-12-24 by akinsey2468