realsenseai/librealsense #5456: Why is there some afterimage in my depth_image?

Dontla, 2019-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description

These afterimages appear to only appear on the left and top of the outline of the object. Do you know how to fix them?

<Describe your issue / question / feature request / etc..>
[image]

[image]
[image]

MartyG-RealSense, 2019-12-16:
Noise on the leftmost side of the image is typically related to a phenomenon called the Invalid Depth Band, which can widen as the camera gets closer to an object.  It is described on pages 59-60 of the current edition of the data sheet document for the 400 Series cameras.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

Dontla, 2019-12-16:
@MartyG-RealSense Do you know which function of pyrealsense can fix it, if there is no ready-made function to call, are there any examples?
Best Regard!

MartyG-RealSense, 2019-12-16:
I thought very carefully about your question.  In the past, some users have taken the approach of removing invalid depth pixels from the image by converting them to a value of zero.

I wonder if trying to use post-processing filters to adjust invalid pixels may be easier.

https://dev.intelrealsense.com/docs/post-processing-filters

For example, the documentation says that the Decimation filter "provides some hole filling capability, as the filter uses valid (non-zero) pixels only".

Dontla, 2019-12-19:
Thank you i got it~

closed 2019-12-26 by RealSenseCustomerSupport