realsenseai/librealsense #5457: auto exposure mode D435

Andre12193, 2019-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|-------------------------------|-------------------------------------------|
| Camera Model                       | D400  | 
| Firmware Version                   | Realsense Viewer V2.29.0 | 
| Operating System & Version  |   Win (8.1/10) | 
| Platform                                 | PC |
| Language                               |  python | 

### Issue Description

Hey guys,

I want to know how the automatic exposure mode works.
 
So, what are the criteria of these mode to set the exposure time?
 
Does a formula or similar exist to understand the settings?
And the last question. We can control the exposure time in the range of 41 to 10000 (Datasheet: Intel Realsense D400 Series Product Family p.67).
Which time does this correspond to in ms and does this range have a linear resolution? 

best regards

André

g2-bernotas, 2019-12-17:
@RealSenseCustomerSupport Also, please explain what the mean intensity range corresponds to. I believe it ranges from 0 to 4096.

RealSenseCustomerSupport, 2020-01-04:
----------------------------------------------

Hi

1) The manual exposure slide bar value means the Exposure Time in unit of "0.1ms", so "10000" means 1000 ms. "41" means 4.1ms (actually 3.9ms). However, the exposure register has 20 bits and each bit represents 1 Line Time, which is the T_Line=30us (RGB sensor).

"10000"=0x0824C(register value)=33,356 x30us=1000ms

      "41"=0x00082(register value)=130 x30us=3.9ms

2) The RGB Auto Exposure is controlled by a 3rd party ISP chip and we don't own the algorithm.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

closed 2020-02-01 by RealSenseCustomerSupport