realsenseai/librealsense #5458: How to save and load depth intrinsic data ?

Uvindu, 2019-12-16:
I use pyrealsense2 to get the depth and color image to a realsense connected device(Rspberry pi). Then I transmit those images to my laptop using ROS(Robot operating System) publisher nodes since the Raspberry pi could not handle high computational algorithms. 
I want to use "pyrealsense2.rs2_deproject_pixel_to_point" in my laptop but it needs depth intrinsic data. 
Is there a method to transmit those data via a ROS custom message or is there any way to save those intrinsic parameters into an offline file then load it on my laptop?

Thank you

ev-mp, 2019-12-17:
@Uvindu , since this is package-specific request that it is probably better to redirect it to the ROS repository - https://github.com/IntelRealSense/realsense-ros/issues

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Hi Uvindu,

If you still need assistance with this please re-open this issue in the realsense-ros repo referenced below.

If we don’t hear from you in 7 days, the issue in this repo will be closed.

Thanks

closed 2020-05-08 by RealSenseCustomerSupport