realsenseai/librealsense #5459: What method can repair the black hole in the depth_image?

Dontla, 2019-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In the process of using the depth map, I found that there are some black holes in the depth map. The data value displayed is 0. In pyrealsense, which function can be used to repair it?

MartyG-RealSense, 2019-12-16:
The post-processing filters that I linked you to in another message can be used for hole-filling too.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

kafan1986, 2019-12-16:
@Dontla 
1) Device preset (Use the appropriate high fill rate JSON preset)
2) Postprocessing filters (decimation+holefilling)

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

Hi @Dontla Do you need further help on this?
Thanks,

closed 2020-01-08 by RealSenseCustomerSupport