realsenseai/librealsense #5460: C# RGB image processing - object detection in RGB image and pixel position mapping back to point cloud vertex

AOndracek, 2019-12-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 } 
| Firmware Version                   | (Open RealSense Viewer --> Click info) 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |                              
| Platform                                 | PC
| SDK Version                            |  { legacy / 2.22 }                          
| Language                            |  {C# }                          
| Segment			|  others }                          | 

### Issue Description
I'd like to get the color frame from the D415 camera in a form of some basic datatype to e.g. detect circles in c# and then map those detected positions to 3D (to a point cloud vertex).

Could you please point me in the right direction with some sample code?
I'm missing documentation for the c# wrapper.
Thank you.

dorodnic, 2019-12-16:
Probably [c# example 3 - processing](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-3-processing/Window.xaml.cs) is a good start, as far as camera management is concerned. 
C# wrapper is less comprehensive than its C++ and python counterparts, but the API is mostly the same across languages so it may be useful to look at other C++ examples

AOndracek, 2019-12-18:
I tried running the example3, but I keep getting System.BadImageFormatException.
(when calling "Context" constructor). Everything seems to be compiled for x86... (only cs-tutorial-3-processing, realsense2 and intel.RealSensed are of importance, right?)
Would  you please have some suggestions?

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@AOndracek Are you running the example on x86 platform?

AOndracek, 2020-01-07:
No, Windows 10, 64 bit...

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

This question may be related to the following thread:
https://github.com/IntelRealSense/librealsense/issues/2125
Generate the solution and set the target platform to x64.

** Please be informed that this ticket will be closed if there we have not heard from you in 7 days. Thank you for your understanding.

closed 2020-04-15 by RealSenseCustomerSupport