realsenseai/librealsense #5461: Wrong runtime path when installing in custom user folder (non sudo)

claudiofantacci, 2019-12-16:
When installing `librealsense` compiled by source on Debian in a custom _user folder_ (sudo not required) the runtime path is not properly set
```
-- Installing: <path>/bin/realsense-viewer
-- Set runtime path of "<path>/bin/realsense-viewer" to ""
```
as a result `ldd realsense-viewer` reports
```
ldd <path>/bin/realsense-viewer
        ...
	librealsense2-gl.so.2.31 => not found
	librealsense2.so.2.31 => not found
        ...
```
and running `realsense-viewer` fails with the following error:  
`
realsense-viewer: error while loading shared libraries: librealsense2-gl.so.2.31: cannot open shared object file: No such file or directory
`
From an early investigation, it seems that `rpath` is not configured. Is this a design choice?
Otherwise, you must either set `LD_LIBRARY_PATH`, which I personally want to always avoid, or you can follow this [CMake RPATH guidelines](https://gitlab.kitware.com/cmake/community/wikis/doc/cmake/RPATH-handling) that are very useful and actually fix the problem. By adding the **full RPATH** support at the beginning of the main `CMakeLists`, i.e.
```cmake
set(CMAKE_SKIP_BUILD_RPATH  FALSE)
set(CMAKE_BUILD_WITH_INSTALL_RPATH FALSE)
set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib")
set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
list(FIND CMAKE_PLATFORM_IMPLICIT_LINK_DIRECTORIES "${CMAKE_INSTALL_PREFIX}/lib" isSystemDir)
if("${isSystemDir}" STREQUAL "-1")
   set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib")
endif()
```
everything then works just fine.

dorodnic, 2019-12-16:
Hello @claudiofantacci 
Thank you for the report. Personally, I was not aware of this cmake capability. IMHO, current status is not by design (@ev-mp ?). We will take a closer look and try to fix this issue. 
Alternatively, feel free to open a pull-request, if you would like attribution for proposed solution.

claudiofantacci, 2019-12-17:
Hi @dorodnic and thanks for reaching out quickly. Let’s see what @ev-mp says about this and in case I can design a PR to get this done.

ev-mp, 2019-12-17:
@claudiofantacci hello, actually we had RPATH configuration in CMakeFiles about two years ago, but then we removed during the work on Debian packaging.
The main reason is that adding RPATH adds hard-coded paths values into the binaries, which makes them susceptible to misbehavior in case of simultaneous usage of multiple versions. Therefore, though the usage of RPATH is not prohibited it is highly discouraged by Debian community.
See the [following page](https://wiki.debian.org/RpathIssue) for their rationale explained.

claudiofantacci, 2019-12-17:
Hey @ev-mp, thanks for the pointing me to the Debian community page.
When packaging a library for Debian release, RPATH should be turned off.
However, by turning off RPATH entirely you force users to install their library in system paths (with sudo), which potentially pollute their system, or by setting env variables like `LD_LIBRARY_PATH`, which is generally discouraged.

I think that RPATH should be set and adapted on the system at use. If you need to package it for Debian release through repository, you can set ```CMAKE_SKIP_RPATH=TRUE```. Another option would be to have RPATH OFF by default and allowing users to turn it on when they want to install it on custom directories or to run everything from the build folder. There are quite some options and I think is reasonable to use them :smile: My experience so far is to have it on and turned off when packaging it for distributions like Debian, brew on macOS, etc. Anyway, this is a design choice from developers so I'm willing to help or test, let me know what you think! :+1:

The code snippet I wrote above should be something easy for you to test if you want :-)

ev-mp, 2019-12-17:
@claudiofantacci hello,
I see no reason not to add the additional granularity level to allow for special cases as the one you've elaborated. But I'd prefer that this would be transparent for most users, i.e. leaving the default w/o RPATH and enabling it on demand via cmake config.
For most users the choice of "/usr/lib/ or "usr/local/lib" is still the de-facto _modus operandi_  as it is aligned with the mainstream debian practices, so introducing non-standard paths shall be made with non-default configuration.

claudiofantacci, 2019-12-18:
Hey @ev-mp thanks for sharing your thoughts
> I see no reason not to add the additional granularity level to allow for special cases as the one you've elaborated.

That's great :rocket: 

> But I'd prefer that this would be transparent for most users, i.e. leaving the default w/o RPATH and enabling it on demand via cmake config. For most users the choice of "/usr/lib/ or [...]

Reasonable and a imho a developer design choice.
In this case the solution provided by CMake guidelines should already comply with your requirements:
```cmake
set(CMAKE_SKIP_BUILD_RPATH  FALSE)
set(CMAKE_BUILD_WITH_INSTALL_RPATH FALSE)
set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
list(FIND CMAKE_PLATFORM_IMPLICIT_LINK_DIRECTORIES "${CMAKE_INSTALL_PREFIX}/lib" isSystemDir)
if("${isSystemDir}" STREQUAL "-1")
   set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib")
endif()
``` 
What it does is:
- uses RPATH for compiled targets in the build directory; this is already the default behaviour
- uses RPATH for compiled targets in the install directory, unless it is a implicit link directory, e.g. `/usr/lib/`, `usr/local/lib`, etc.
As I don't know all the platform you're supporting, CMake versions, etc., you might want to test this :-)

Otherwise we can play around with options and cache variables to enforce your default behaviour and allow the users to tweak it for their need.

RealSenseCustomerSupport, 2020-04-03:
----------------------------------------------

Hi @claudiofantacci,

We are reviewing your change proposal request and will update you when the decision is made.

Thank you!

RealSenseCustomerSupport, 2020-06-16:
----------------------------------------------

@claudiofantacci Have you submitted your PR? If yes, please share the PR link to us and we'll look into it. Thanks!

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@claudiofantacci  Could you please update? Thanks!

claudiofantacci, 2020-06-22:
@RealSenseCustomerSupport I did not submit about this as I currently can't do it. Sorry about that. Having said that, the [snippet code here](https://github.com/IntelRealSense/librealsense/issues/5461#issuecomment-567056430) can be just used as-is in the main `CMakeLists` file as explained in the [CMake RPATH guidelines](https://gitlab.kitware.com/cmake/community/wikis/doc/cmake/RPATH-handling).

RealSenseSupport, 2020-06-29:
@claudiofantacci  Thanks for your input!  We'll have our engineering team to evaluate it. And any other questions about this?

closed 2024-03-29 by MartyG-RealSense