realsenseai/librealsense #5463: Combine D415 with T265 question

gerokatseros, 2019-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

gerokatseros, 2019-12-16:
I have both T265 and D415 and I wonder how can I collect the pointcloud from D415 from one pose (I found documentation about pointscloud in this forum) and then append new one on this taking into account T265 translation and rotation from new position. I want to do this in python but haven't found something relative in the forum.

MartyG-RealSense, 2019-12-16:
Intel have published a guide for using the T265 and 400 Series depth cameras together.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

The document features a link to a ROS package that can be used to generate a 2D occupancy map based on depth images and poses from depth and tracking cameras respectively.  Here is the direct link to the package for your convenience.  I'm not sure it is exactly what you were looking for but may provide useful guidance.

https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping/occupancy`1

gerokatseros, 2019-12-17:
thank you for your quick reply ... second link doesn't seem to work

gerokatseros, 2019-12-17:
I was looking more for some python examples about it to build my own script

MartyG-RealSense, 2019-12-17:
I have edited the second link to work.  Sorry about that.

I searched very carefully for a long time for a Python example for combining depth and tracking but could not find one unfortunately.  The question has come up a couple of times before, such as in the case linked to below:

https://forums.intel.com/s/question/0D50P00004LVODx/how-do-i-transform-a-d435-point-cloud-using-the-pose-from-a-t265-tracking-camera-in-python?language=en_US

The link below describes using ROS with a program called Rviz to update a point cloud based on the orientation of the T265.

https://github.com/IntelRealSense/realsense-ros/issues/1011#issuecomment-562078935

isirana, 2020-01-03:
I have the same issue when I combine D435i and T265 cameras together to get the point cloud...

MartyG-RealSense, 2020-01-03:
@isirana  Is your dad issue that you are looking for a Python script for using D435i and T265 together please?

isirana, 2020-01-03:
> @isirana Is your same issue that you are looking for a Python script for using D435i and T265 together please?

yes, it is the same issue.... I have tried different ways, but I couldn't solve it.

MartyG-RealSense, 2020-01-03:
I found a case with a Python script where the user was trying to use D435 and T265 together with Python.  They posted their script, and a member of the Intel RealSense team made a code-fix comment below it that the user said made the script work successfully.

https://github.com/IntelRealSense/librealsense/issues/3697#issuecomment-483089732

isirana, 2020-01-03:
> I found a case with a Python script where the user was trying to use D435 and T265 together with Python. They posted their script, and a member of the Intel RealSense team made a code-fix comment below it that the user said made the script work successfully.
> 
> [#3697 (comment)](https://github.com/IntelRealSense/librealsense/issues/3697#issuecomment-483089732)

Thanks, I will try to do that

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi @gerokatseros

I think this thread can be closed since a lot of good information has been provided by MartyG-RealSense.  Let us know if anything else is needed on this thread.

closed 2020-03-17 by RealSenseCustomerSupport