realsenseai/librealsense #5464: Set laser power for D415 vis ROS launch argument?

scottnortman, 2019-12-17:
Is it possible to set the D415 laser power via a ROS launch file argument?  I can not find an example of this. 

Thank you,
Scott

dorodnic, 2019-12-17:
@doronhi

doronhi, 2019-12-17:
It is not possible via the launch file but it is using the json file.
See my response [here](https://github.com/IntelRealSense/realsense-ros/issues/981)
The line in the json file you need to modify is:
    "controls-laserpower": "200"

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

Hi @scottnortman Do you need further help on this?
Thanks,

scottnortman, 2019-12-23:
No, I figured it out via the command line dynamic_reconfigure call; thank you.

closed 2019-12-23 by scottnortman

doronhi, 2019-12-29:
@scottnortman , Thanks for your update.
I was wrong and misleading.
If you can change the parameter using dynamic_reconfigure then you can change it in launch file.
In that case, add the following lines to your launch file:
```
  <rosparam>
      /camera/stereo_module/laser_power: 200
  </rosparam>
```