realsenseai/librealsense #5466: RealSense D415 Multi Camera Record is not working

wodndb, 2019-12-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.31.0   | 
| Language                            |  Unity / C# | 
| Segment			| ect. | 

### Issue Description

I updated RealSense SDK 2.0 from 2.24.0 to 2.31.0. Unfortunatly I can't record bag with 3 RealSense Camera.

Just one RsDevice run, others not run...

Should I downgrade RealSense SDK 2.0?

[image]

[image]

<Test Scene Images, I added serial number in live mode>

```
ExternalException: rs2_pipeline_wait_for_frames(pipe:000001C90CA61680)
Rethrow as Exception: Frame didn't arrived within 5000
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
Intel.RealSense.Pipeline.WaitForFrames (System.UInt32 timeout_ms) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
RsDevice.WaitForFrames () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:159)
System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ThreadHelper.ThreadStart () (at <567df3e0919241ba98db88bec4c6696f>:0)
UnityEngine.UnhandledExceptionHandler:<RegisterUECatcher>m__0(Object, UnhandledExceptionEventArgs)
Intel.RealSense.NativeMethods:rs2_pipeline_wait_for_frames(IntPtr, UInt32, Object&)
Intel.RealSense.Pipeline:WaitForFrames(UInt32)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:159)
System.Threading.ThreadHelper:ThreadStart()

```

```
ExternalException: rs2_pipeline_wait_for_frames(pipe:000001C9688DD0C0)
Rethrow as Exception: Frame didn't arrived within 5000
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
Intel.RealSense.Pipeline.WaitForFrames (System.UInt32 timeout_ms) (at <cc4cf66ca4ac447089d4777657d39da9>:0)
RsDevice.WaitForFrames () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:159)
System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <567df3e0919241ba98db88bec4c6696f>:0)
System.Threading.ThreadHelper.ThreadStart () (at <567df3e0919241ba98db88bec4c6696f>:0)
UnityEngine.UnhandledExceptionHandler:<RegisterUECatcher>m__0(Object, UnhandledExceptionEventArgs)
Intel.RealSense.NativeMethods:rs2_pipeline_wait_for_frames(IntPtr, UInt32, Object&)
Intel.RealSense.Pipeline:WaitForFrames(UInt32)
RsDevice:WaitForFrames() (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:159)
System.Threading.ThreadHelper:ThreadStart()

```

<Error code in console>

wodndb, 2019-12-17:
Please Modify RsConfiguration.cs in Unity Package!
This issue was found when I use RealSense SDK 2.0 ver 2.24.0...

```
            case Mode.Record:
                cfg.EnableDevice(RequestedSerialNumber);   // Add this!
                foreach (var p in Profiles)
                    cfg.EnableStream(p.Stream, p.StreamIndex, p.Width, p.Height, p.Format, p.Framerate);
                if (!String.IsNullOrEmpty(RecordPath))
                    cfg.EnableRecordToFile(RecordPath);
                break;
```

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

wodndb, 2020-12-07:
I don't need help with this. I just want that someone fix this codes and commit.. Thanks.

closed 2020-12-07 by wodndb