realsenseai/librealsense #5469: File download hash mismatch

AOndracek, 2019-12-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   {Win (10)
| Platform                                 | PC
| SDK Version                            |  { 2.0 }                          | 

### Issue Description
When I clone the repository and run CMake (cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_SHARED_LIBS=ON), I keep getting hash mismatch error(below).
I tried downloading/compiling multiple times with always the same result.
Earlier versions (e.g. since 2.22) compiled fine.
What could be the cause? 
Thank you.

CMake Error at third-party/libtm/fw/CMakeLists.txt:22 (file):
  file DOWNLOAD HASH mismatch

    for file: [D:/SoftwareMix/RealSenseLib/librealsense/build/third-party/libtm/fw/target-0.2.0.879.mvcmd]
      expected hash: [b7b722b785fcf5f8c31877173c8ba02fed120e03]
        actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
             status: [22;"HTTP response code said error"]

Call Stack (most recent call first):
  third-party/libtm/fw/CMakeLists.txt:47 (target_binary)

ev-mp, 2019-12-17:
@AOndracek , iirc the hash `da39a3ee5e6b4b0d3255bfef95601890afd80709` corresponds to zero-sized file, which can mean that the download has failed, file being locked or access permissions not valid.
Try to verify the internet connection (proxies, firewall), then remove all the intermediate files, including the one mentioned above, and make a clean cmake build.

AOndracek, 2019-12-18:
Ok, the file size is 0, it doesn't seem to be locked or with restricted access permissions. I ran the cmake from cmd "as admin" as well, but to no avail. There is also no proxy. I'm not sure how to check the internet connection however - mainly the firewall, could you give me some advice on that please?

ev-mp, 2019-12-18:
Direct link http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/TM2/FW/target/0.2.0.879/target-0.2.0.879.mvcmd

AOndracek, 2019-12-18:
Thank you very much - while using the link, I discovered that there was a HTTP filter I didn't know about,  which was blocking the download. 
So now it is solved.

closed 2019-12-18 by AOndracek

becker666, 2024-05-25:
I get the same problem but the file is a non-empty  png which i can load in preview fine

MartyG-RealSense, 2024-05-26:
Hi @becker666  There is further advice about the **file DOWNLOAD HASH mismatch** error at https://github.com/IntelRealSense/librealsense/issues/4833#issuecomment-529836674