realsenseai/librealsense #5470: T265: Stream outages over several seconds

neilyoung, 2019-12-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

I made several rounds in larger venues. T265 mounted to the macbook pro. Proper USB cables used. Also after replacement no change. 

I'm using the realsense-viewer app and it can be pretty good seen, how the video and all IMU streams are stuck for up to 5 seconds. With the original cable the outages were not that long, but there were.

Not sure if it is related, but this is the console output of the last run:

```
 $ realsense-viewer 
objc[28269]: Class GLFWLayoutListener is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7cf8) and /usr/local/bin/realsense-viewer (0x10f2bf420). One of the two will be used. Which one is undefined.
objc[28269]: Class GLFWWindowDelegate is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7d20) and /usr/local/bin/realsense-viewer (0x10f2bf448). One of the two will be used. Which one is undefined.
objc[28269]: Class GLFWApplicationDelegate is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7d98) and /usr/local/bin/realsense-viewer (0x10f2bf4c0). One of the two will be used. Which one is undefined.
objc[28269]: Class GLFWContentView is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7dc0) and /usr/local/bin/realsense-viewer (0x10f2bf4e8). One of the two will be used. Which one is undefined.
objc[28269]: Class GLFWWindow is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7e38) and /usr/local/bin/realsense-viewer (0x10f2bf560). One of the two will be used. Which one is undefined.
objc[28269]: Class GLFWApplication is implemented in both /usr/local/lib/librealsense2-gl.2.31.0.dylib (0x10f4f7e60) and /usr/local/bin/realsense-viewer (0x10f2bf588). One of the two will be used. Which one is undefined.
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 181941166: [0x/1:386] TM2 version 0.2.0.896, sku 2155
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 349339991: [0x/29:190] LeonIPCRxReassignSinkThread status: 0
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 349445026: [0x/29:192] LOS started
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 349514050: [0x/29:193] LeonIPC message size is 1
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 91746166: [0x/29:190] LeonIPCRxReassignSinkThread status: 0
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 91890948: [0x/29:192] LRT started
 17/12 14:52:49,009 WARNING [0x70000ee5b000] (tm-device.cpp:1386) T265 FW message: 91969535: [0x/29:193] LeonIPC message size is 1
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Motion
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Video
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Roi
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Depth Frame
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Advanced Mode
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Video Profile
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Disparity Frame
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Motion Profile
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Pose Frame
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Pose Profile
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Tm2
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Software Device
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Software Sensor
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Decimation Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Threshold Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Disparity Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Spatial Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Temporal Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Hole Filling Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Zero Order Filter
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Pose
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Pose Sensor
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Wheel Odometer
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Global Timer
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Updatable
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Update Device
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Tm2 Sensor
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Auto Calibrated Device
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Motion
 17/12 14:54:11,583 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Video
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Roi
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Depth Frame
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Advanced Mode
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Video Profile
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Disparity Frame
 17/12 14:54:11,584 WARNING [0x70000e837000] (record_device.cpp:257) Extensions type is unhandled: Motion Profile
[... 92 lines cut ...]
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Hole Filling Filter
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Zero Order Filter
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Pose
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Pose Sensor
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Wheel Odometer
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Global Timer
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Updatable
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Update Device
 17/12 15:09:51,252 WARNING [0x700002781000] (record_device.cpp:257) Extensions type is unhandled: Tm2 Sensor
[... cut ...]
 17/12 15:16:54,871 WARNING [0x700002575000] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: Operation timed out, err. num: 60
 17/12 15:16:54,871 ERROR [0x700002575000] (tm-device.cpp:1978) Bulk request error RS2_USB_STATUS_NO_DEVICE
 17/12 15:17:32,131 WARNING [0x1102e8dc0] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: Operation timed out, err. num: 60
 17/12 15:17:32,131 ERROR [0x1102e8dc0] (tm-device.cpp:1978) Bulk request error RS2_USB_STATUS_NO_DEVICE
~ $ 

```

MartyG-RealSense, 2019-12-17:
Another user who had these unhandled extension types during recording recently was advised that they are "expected and not indicative of misbehaviour".

https://github.com/IntelRealSense/librealsense/issues/5006

neilyoung, 2019-12-18:
@MartyG-RealSense Nice find, but I'm not that much concerned about the exceptions. I'm concerned about the outages in the stream. I can't believe, that stream outages, wich affect the video **and** the pose stream, are "desired behaviour". It's like letting drive the robot with someone in the front passenger seat who covers the robot's eyes for a few seconds from time to time. Somehow I can't imagine that that's what you want to do. I now check whether these gaps also apply to the "pose only" stream. If so, this is a serious problem for the T265.

MartyG-RealSense, 2019-12-18:
It wasn't desired behavior, but 'expected' behavior. 

A debugging rule of thumb that I use is to look at the first occurrence of an error in a log, since an error may cause a cascade of subsequent areas in other areas that would not be erroring if the initial error had not occurred (i.e focusing on the first error helps to avoid investigating in areas where there may not be a problem).

If we ignore the extension handling warnings because they are 'expected', the next "real" error after that is the mempool error.

WARNING [0x7000024f2000] (tm-device.cpp:1386) T265 FW message: 434145045981: [0x
/5:100] mempool out of memory: video

In both occasions when this error appears, it is immediately followed by time-out errors.  So it might be reasonably expected that the mempool error is the trigger for timeouts.

bfulkers-i, 2019-12-18:
Thanks @neilyoung, and also great to see your tests outdoors in #5472. The data outages are definitely not expected. I added a recent patch to #5213 yesterday which may help with this:  
64c753b. I'm not sure that's what you are experiencing here, but it feels like it might be related. You may also try enabling only the pose stream (maybe you already are) to reduce the amount of data that needs to be processed on your host machine.

neilyoung, 2019-12-18:
@MartyG-RealSense Cool, no problem with your wording :) I probably misunderstood your sentence.

@bfulkers-i Great, will patch soon. As you said: I already enabled just the pose stream but the results are not too encouraging. I will report later in the original thread  https://github.com/IntelRealSense/librealsense/issues/5472. Thanks for now

radfordi, 2020-01-08:
@neilyoung, this indicates a firmware issue:
> mempool out of memory: video

neilyoung, 2020-01-08:
Ok. Then there is hope :)

neilyoung, 2020-01-09:
> @bfulkers-i Great, will patch soon. As you said: I already enabled just the pose stream but the results are not too encouraging. I will report later in the original thread #5472. Thanks for now

While reviewing this I found, that this statement is not true. Using only the pose stream did not show any outage. I tested it later and these are the results https://github.com/IntelRealSense/librealsense/issues/5472#issuecomment-567010861

Very accurate in timing, but drifty :)

radfordi, 2020-01-09:
@neilyoung, were you recording the images when you got the drops in #5472?  Is so, then the drops are likely caused by writing the data to SSD without sufficient bandwidth.   You have to be very careful when recording to avoid this issue.

radfordi, 2020-01-09:
@neilyoung, the drifting is expected on loops this large.   I have two questions here.   Do we run out of internal map, and if not, do we close the loop, i.e. are there relocalizations?

neilyoung, 2020-01-09:
Yes. Video was on. And the Mac was running with lis closes, but this was not the issue.. I could reproduce with lis open. Maybe I try with a ram disk next time.

neilyoung, 2020-01-31:
closed for lack of interest

closed 2020-01-31 by neilyoung