realsenseai/librealsense #5472: T265: Precision of the tracking outdoors (or at large scale)

neilyoung, 2019-12-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

After having achieved pretty good results with the T265 indoors (even in a big mall), I was trying to do some tests walks outdoors today. And the results are not that superior anymore. I made an "8" around the block and projected the results to WGS84, into a KML, which I opened in Google Earth then.

Below you find images of the walked path (red) and the derived path.

[image]

[image]

The gaps in the derivation are really gaps in the input stream. I recorded the walk using RealSense Viewer. At least on MacOS, with the original cable, there are outages, which can be seen in the Viewer: The video is stuck, the IMU readings too, the poses don't change. 

I treat it as problems with the USB for now.

But what can be done to improve the accuracy? More training? 

As another test I made a straight walk 180 forth, U-turn and 180 back. I will not publish the results: The resulting path was 80 too short....:(

I have all the bag files, the derived video and KML here, if somebody of the team is interested to investigate.

neilyoung, 2019-12-18:
OK, in order to mitigate the possible load problems caused by the fisheye *and* pose stream I today made another walk around the block, now with just the pose stream enabled. The good news first: There are no outages. 200 fps.

[image]

The white straight line represents my real path, the circles are the derived WGS84 coordinates. I started going southwards. And when I returned I walked a part of the way again in order to provoke a loop-closing, crossed the street and returned to the starting point.

The green line represents the delta between these points. It's about 14 m, although it is 0 in reality.

[image]

I have double-checked the results with the realsense-viewer, which is still not able to display a large path (issue https://github.com/IntelRealSense/librealsense/issues/5078): There is neither a loop closing (I would expect to get a relocation indication) nor do start and end match. 

I additionally observed an unclear "-3" in the y coordinate at the end. This is what I'm already have noticed with other runs. The Y is not consistent over time.

The drift is enormous: About 17 m in the southern part, up to 30 in the north-western part, 18 in the northern part and 7 in the eastern part.

neilyoung, 2019-12-18:
However, the path re-construction from the poses is pretty good, there just seems to be an issue with the calibration....

neilyoung, 2019-12-18:
BTW: If somebody could help me to understand, why this attempt to set a notification callback ends up in 

`Could not find requested sensor type!
`

would be nice...

```
        config = rs.config()
        pipeline = rs.pipeline()
        config.enable_device_from_file(args.bagfile, False)
        config.enable_stream(rs.stream.pose)

        pose_sensor = config.resolve(pipeline).get_device().first_pose_sensor()
        pose_sensor.set_notifications_callback(callback)

```

There seems to be a problem with the way I'm obtaining a pose_sensor (which is required for setting/getting static nodes and for setting a callback). See also https://github.com/IntelRealSense/librealsense/issues/5479

neilyoung, 2019-12-18:
Hmm... maybe because there is no device? :) So is there a way to obtain a pose_sensor object from a playback?

radfordi, 2020-01-09:
@neilyoung, the drifting is expected on loops this large, say on the order of 1% of the path length.   Eyeball, I'd say it looks slightly larger than that.

I have two questions  here.   Do we run out of internal map, and if not, do we close the loop,  i.e. are there relocalizations?

neilyoung, 2020-01-09:
@radfordi Hi Jim. I didn’t observe a loop closing, even though I was trying to enforce it. So there was no such an indication. And for the question if the map was overrun, I can’t say much since I’m not aware of any indication of that. But I can repeat the test anytime and probably these days is a good time to do that after all the fixes lately. Will return to you soon.

neilyoung, 2020-01-31:
closed for lack of interest

closed 2020-01-31 by neilyoung