realsenseai/librealsense #5473: T265 crashing and D435 won't stream colour

matildameasday, 2019-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265, D435  | 
| Firmware Version                   | 0.2.0.879,05.11.06.250 | 
| Operating System & Version | Linux (Ubuntu 18.04.3 LTS) | 
| Kernel Version (Linux Only)    |  4.19.76                                    | 
| Platform                                 |Raspberry Pi 4 model B 2GB|
| SDK Version                            |  2.31.0                      | 
| Language                            |  python                         | 
| Segment			|  others (sensor suit)                        | 

### Issue Description
My T265 camera is crashing after a varying amount of time. Before crashing, It is throwing these messages:

    13:39:35.436 [1811] [E] Device-CC20: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) response to host got LIBUSB_ERROR_TIMEOUT. Host did not return answer
    13:39:35.436 [1811] [E] Device-CC20: Error syncing TM2 and Host clocks
    13:39:36.037 [1811] [E] Device-CC20: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_TIMEOUT. Bytes requested 6, bytes transferred 0
    13:39:36.037 [1811] [E] Device-CC20: Error syncing TM2 and Host clocks recieved frame x: 5.00306e-05, y: 0.000228195, z: 0.000429381
    13:39:36.638 [1811] [E] Device-CC20: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_TIMEOUT. Bytes requested 6, bytes transferred 0
    13:39:36.639 [1811] [E] Device-CC20: Error syncing TM2 and Host clocks
    13:39:37.240 [1811] [E] Device-CC20: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_TIMEOUT. Bytes requested 6, bytes transferred 0
    13:39:37.240 [1811] [E] Device-CC20: Error syncing TM2 and Host clocks
    13:39:37.702 [1871] [E] Device-CC20: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
    13:39:37.702 [1870] [E] Device-CC20: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
    13:39:37.903 [1811] [E] Device-CC20: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
    13:39:37.904 [1811] [E] Device-CC20: Entered state [ERROR_STATE]

Then, when calling wait_for_frames() the error is thrown:

    RuntimeError: Frame didn't arrive within 5000

I have the camera connected connected to the raspberry pi via a powered usb hub. I have tried connecting it directly to the pi and with multiple different usb cables.

The testing script I am running it with and getting the error (last time it took about 2 minutes to crash)

    import sys
    sys.path.append("/usr/local/lib")
    import pyrealsense2 as rs
    import time

    t_pipeline = rs.pipeline()
    t_config = rs.config()
    t_config.enable_stream(rs.stream.pose)
    t_pipeline.start(t_config)

    while True:

        t_frame = t_pipeline.wait_for_frames()
        print ("recieved frame")
        pose_frame = t_frame.get_pose_frame()

        pose = pose_frame.get_pose_data()

        print(pose.translation)

        time.sleep(2)

    t_pipeline.stop()

I am also using a D435 camera which was working fine when I was using a previous version of the SDK (2.29). I installed the latest release (2.31), and it now will not stream the RGB. It will stream only depth. For example,

    d_config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
    d_pipeline.start(d_config)
    d_frame = d_pipeline.wait_for_frames()

will run successfully. However,

    d_config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 6)
    d_pipeline.start(d_config)
    d_frame = d_pipeline.wait_for_frames()

will fail at `wait_for_frames()` with the same run-time error (even with lower resolutions),

    RuntimeError: Frame didn't arrive within 5000

To get around this error, I downgraded back to librealsense2 version 2.29, but the T265 error remains.

radfordi, 2020-01-08:
I'm not seeing an error with what will become v2.32.0 when running the above script.  This is presumably started working after we merged #5213 .

RealSenseCustomerSupport, 2020-01-23:
----------------------------------------------

Hi matildameasday,

Can you please confirm this does not occur in our latest release?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.32.1

Thanks

matildameasday, 2020-01-28:
Hi,
I've installed version 2.32.1. The D435 still won't work with the colour stream, and the issue with the T265 remains. I've downgraded back to version 2.29 again.

RealSenseSupport, 2020-09-15:
Thank you for highlighting this issue while using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-21 by RealSenseSupport