realsenseai/librealsense #5474: can't find lib file ande include file after install the package

CrazyChildren, 2019-12-18:
platform: ubuntu 16.04LTS

I try to install the realsense SDK2.0 in ubuntu 16.04LTS, and followed the guide:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

and i have install the package by :
`sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev
sudo apt-get install librealsense2-dbg`

and the guide said that :With dev package installed, you can compile an application with librealsense using g++ -std=c++11 filename.cpp -lrealsense2 or an IDE of your choice

I want use realsense in my Qt program, but i can't find **lib path** and **include path** that should be specified in Qt  or other IDE,

so, if you could give some suggestion that what should i do to use realsense in other IDE or where to find realsense lib, include, bin path in ubuntu.

ev-mp, 2019-12-18:
@CrazyChildren, the installation paths for the SDK Debian packages are standard -
 `/usr/lib`, `/usr/include` and `usr/bin`. Use `dpkg -L librealsense2-dev` for verification.

On the other hand, if the other project you're linking the SDK with uses CMake then to finding the SDK's library should be done with putting `find_package(realsense2)` into the Cmake.

CrazyChildren, 2019-12-18:
> @CrazyChildren, the installation paths for the SDK Debian packages are standard -
> `/usr/lib`, `/usr/include` and `usr/bin`. Use `dpkg -L librealsense2-dev` for verification.
>
> On the other hand, if [...]

thanks a lot!! the command really work!
actually library file(.so) located in `/usr/lib/x86_64-linux-gnu/librealsense2.so` and include path is `/usr/include/librealsense2`

closed 2019-12-18 by CrazyChildren