realsenseai/librealsense #5476: OpenCv Samples Eigen3 not found by Cmake

kromond, 2019-12-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 , D415| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I am following the instructions for the OpenCV kinfu sample.  I have compiled OpenCV with the opencv_contrib without issue.  When using Cmake Gui, Configure has no issues however I am unable to Generate, getting this error all the items in the wrappers/opencv : 

> CMake Error at wrappers/opencv/imshow/CMakeLists.txt:6 (add_executable):
  Target "rs-imshow" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

I have a new install of Eigen3 and OpenCv examples all work as expected.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@kromond Please refer to the link to configure `Eigen3_DIR` cmake's variable.
https://eigen.tuxfamily.org/dox/TopicCMakeGuide.html

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@kromond Did you get chance to try? Looking forward to your reply. Thanks!

kromond, 2020-02-13:
Yes that was the clue I needed.  Thank you very much

closed 2020-02-13 by kromond