realsenseai/librealsense #5479: T265: How to `enable_record_to_file` correctly?

neilyoung, 2019-12-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

I'm having a well working script, basically running the well known sequences:

 ```
  cfg = rs.config()       
  cfg.enable_stream(rs.stream.pose)
  pose_sensor = cfg.resolve(pipe).get_device().first_pose_sensor()
  pose_sensor.set_option(rs.option.enable_map_preservation, 1)

   if args.record:
       cfg.enable_record_to_file("./file.bag")
  
  if args.load_map:
        print(datetime.datetime.now(), "Loading map '{}'".format(args.load_map))
        with open(args.load_map, mode='rb') as map_file:
            map = map_file.read()
            pose_sensor.import_localization_map(map)
    # Start streaming with requested config
    pipe.start(cfg)

```

It then goes into the 

```
   while (True):
         frames = pipe.wait_for_frames(10000)
         ...
````

loop and simply works.

I cannot make the bag recording run. Whenever I enable recording, the script excepts with `No device connected` from different points in the scrpt.

```
Traceback (most recent call last):
  File "pose.py", line 287, in <module>
    main()
  File "pose.py", line 74, in main
    pose_sensor = cfg.resolve(pipe).get_device().first_pose_sensor()
RuntimeError: No device connected

```

or

```
Traceback (most recent call last):
  File "pose.py", line 287, in <module>
    main()
  File "pose.py", line 93, in main
    pipe.start(cfg)
RuntimeError: No device connected
```

What am I supposed to do or not to do in order to enable recording?

BTW: Somethingi is all the time created on my HD:

`
-rw-r--r--     1 decades  staff         4707 18 Dez 10:22 the_file.bag
`

neilyoung, 2019-12-18:
I tried this script, which I found here https://github.com/IntelRealSense/librealsense/issues/2172 and which found to be OK by @RealSense-Customer-Engineering , slightly modified of course, and it works.

```
import pyrealsense2 as rs
import time

pipeline_1 = rs.pipeline()
config_1 = rs.config()
config_1.enable_stream(rs.stream.pose)
config_1.enable_record_to_file('test1.bag')

pipeline_1.start(config_1)
start = time.time()

try:
	while True:
		frames = pipeline_1.wait_for_frames()
		print (time.time()-start)
finally:
	pipeline_1.stop()
```

So the difference it the call to obtain a `pose_sensor`. If I omit this, everything is fine. I'm getting poses and the file is recorded...

Any thoughts?

neilyoung, 2019-12-18:
It is really a matter of the two calls, which seem to be mutually exclusive currently

`pose_sensor = config.resolve(pipeline).get_device().first_pose_sensor()
`

and

 `config.enable_record_to_file("./file.bag")
`

Calling both - start refuses with "no device connected"
Calling one of each exclusively let's me either load/save maps, set/get static nodes _or_ record to file.

Not both together... This deserves a closer look :)

bfulkers-i, 2019-12-19:
Thanks @neilyoung, this exposed a bug (which I fixed in 351ac9832fc0be8348718a1fa3d31fa810134372) and a missing piece of documentation which @dorodnic will probably address in the future with example code or more documentation.

I believe this very slightly modified version of your example code should work:
```
import pyrealsense2 as rs

pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.pose)
if args.record:
  cfg.enable_record_to_file("./file.bag")

# Do not call cfg.enable_* after cfg.resolve has been called
pose_sensor = cfg.resolve(pipe).get_device().first_pose_sensor()
pose_sensor.set_option(rs.option.enable_map_preservation, 1)

if args.load_map:
   print(datetime.datetime.now(), "Loading map '{}'".format(args.load_map))
   with open(args.load_map, mode='rb') as map_file:
       map = map_file.read()
       pose_sensor.import_localization_map(map)
# Start streaming with requested config
pipe.start(cfg)
```

The key issue is that you must call all of your `cfg.enable_` functions before you call `cfg.resolve`. `cfg.resolve` takes all of your requested input and tries to match it with a set of devices, which then get used by the pipeline (this is why it is important to call `pipe.start` with the config). If you call `cfg.enable_` after `cfg.resolve_` it makes your sensor invalid and so you lose all the configuration you've already done.

neilyoung, 2019-12-19:
@bfulkers-i Thanks, indeed. That fixed it. But there is a new issue. 

I'm pretty often catching `Bus Error 10` with this little test script:

```
import pyrealsense2 as rs
import time

def cb(notification):
    print(notification)

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.pose)
config.enable_record_to_file('test1.bag')

device = config.resolve(pipeline).get_device()

print("Firmware:", device.get_info(rs.camera_info.firmware_version))
print("Serial Number:", device.get_info(rs.camera_info.serial_number))
print("USB:", device.get_info(rs.camera_info.usb_type_descriptor))

# Get first pyrealsense2.sensor in order to set notification callback
sensor = device.sensors[0]
sensor.set_notifications_callback(cb)
# Get first pyrealsense2.pose_sensor in order to set options, load maps, save maps, set static node, get static node
pose_sensor = device.first_pose_sensor()
pose_sensor.set_option(rs.option.enable_map_preservation, 1)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        pose = frames.get_pose_frame()
        result, translation, rotation = pose_sensor.get_static_node("test")

finally:
    pipeline.stop()

```

Basically every 2nd attempt fails like so.

neilyoung, 2019-12-30:
Still present in 0.2.0.908, #5504 and probably based upon the same root problem as https://github.com/IntelRealSense/librealsense/issues/5554

radfordi, 2020-01-08:
Thanks @neilyoung.  I can reproduce this and your right, "every second run fails".

radfordi, 2020-01-08:
This is crashing when we try and dispatch a relocalization notification:
```
frame #21: 0x0000000104938968 librealsense2.2.32.dylib`void dispatcher::invoke<librealsense::record_device::write_notification(unsigned long, librealsense::notification const&)::$_8>(this=0x0000000102347098, item=_Target @ 0x00007000069596b0, is_blocking=false)::$_8, bool) at concurrency.h:256:32
   253              if(is_blocking)
   254                  _queue.blocking_enqueue(std::move(item));
   255              else
-> 256                  _queue.enqueue(std::move(item));
   257          }
   258      }
   259
(lldb) p item
(_Target) $0 = {
  this = 0x00000001023482a0
  sensor_index = 0
  capture_time = (__rep_ = 477650515)
  n = {
    category = RS2_NOTIFICATION_CATEGORY_POSE_RELOCALIZATION
    type = 0
    severity = RS2_LOG_SEVERITY_INFO
    description = "T2xx: Relocalization occurred. id: 1, timestamp: 1.57851e+12 ms"
    timestamp = 1578514356662.4263
    serialized_data = ""
  }
}
```

neilyoung, 2020-01-13:
If it helps, here is a crash dump, taken on my Mac. It shows, that something is screwing up in the notification corner...

<details>
  <summary>Click to expand</summary>
<pre>

 Process:               Python [5425]
Path:                  /usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               3.7.5 (3.7.5)
Code Type:             X86-64 (Native)
Parent Process:        bash [951]
Responsible:           Terminal [475]
User ID:               501

Date/Time:             2019-12-29 14:07:05.915 +0100
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        4C2D26AD-F99F-6E44-AA0B-AD3BC0775946

Time Awake Since Boot: 3800 seconds

System Integrity Protection: enabled

Crashed Thread:        10

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000700006133ff8
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Bus error: 10
Termination Reason:    Namespace SIGNAL, Code 0xa
Terminating Process:   exc handler [5425]

VM Regions Near 0x700006133ff8:
    Stack                  00007000060b1000-0000700006133000 [  520K] rw-/rwx SM=COW  thread 9
--> STACK GUARD            0000700006133000-0000700006134000 [    4K] ---/rwx SM=NUL  stack guard for thread 10
    Stack                  0000700006134000-00007000061b6000 [  520K] rw-/rwx SM=COW  thread 10

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff699ec5be __select + 10
1   corecext.cpython-37m-darwin.so	0x000000010ea989d2 select_poll + 162
2   corecext.cpython-37m-darwin.so	0x000000010ea899c1 ev_run + 1617
3   corecext.cpython-37m-darwin.so	0x000000010ea9b43d __pyx_pw_6gevent_5libev_8corecext_4loop_15run + 605
4   org.python.python             	0x000000010db644bd _PyMethodDef_RawFastCallKeywords + 537
5   org.python.python             	0x000000010db68cbd _PyMethodDescr_FastCallKeywords + 81
6   org.python.python             	0x000000010dbf83ea call_function + 782
7   org.python.python             	0x000000010dbf1312 _PyEval_EvalFrameDefault + 6742
8   org.python.python             	0x000000010db63dec function_code_fastcall + 106
9   org.python.python             	0x000000010db6475f _PyObject_Call_Prepend + 131
10  org.python.python             	0x000000010db63b25 PyObject_Call + 136
11  greenlet.cpython-37m-darwin.so	0x000000010ec20b20 g_initialstub + 1200
12  greenlet.cpython-37m-darwin.so	0x000000010ec2031a g_switch + 346

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff699e6ce6 __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff69aa8185 _pthread_cond_wait + 701
2   libc++.1.dylib                	0x00007fff669bfcc1 std::__1::condition_variable::__do_timed_wait(std::__1::unique_lock<std::__1::mutex>&, std::__1::chrono::time_point<std::__1::chrono::system_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > >) + 93
3   librealsense2.2.31.0.dylib    	0x000000010f0b5556 single_consumer_queue<std::__1::function<void (dispatcher::cancellable_timer)> >::dequeue(std::__1::function<void (dispatcher::cancellable_timer)>*, unsigned int) + 262 (__mutex_base:419)
4   librealsense2.2.31.0.dylib    	0x000000010f0b5348 dispatcher::dispatcher(unsigned int)::'lambda'()::operator()() const + 104 (concurrency.h:225)
5   librealsense2.2.31.0.dylib    	0x000000010f0b526d void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, dispatcher::dispatcher(unsigned int)::'lambda'()> >(void*) + 45 (memory:2649)
6   libsystem_pthread.dylib       	0x00007fff69aa7e65 _pthread_start + 148
7   libsystem_pthread.dylib       	0x00007fff69aa383b thread_start + 15

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff69aa3818 start_wqthread + 0

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff699e4362 swtch_pri + 10
1   libsystem_pthread.dylib       	0x00007fff69aa539d sched_yield + 11
[... 912 lines cut ...]
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       12K        3 
===========                     =======  ======= 
TOTAL                            804.6M      678 
TOTAL, minus reserved VM space   803.1M      678 

Model: MacBookPro11,2, BootROM 157.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2,6 GHz, 16 GB, SMC 2.18f15
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1435.3)
Bluetooth: Version 7.0.2f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: TS480GJDM820, 480,1 GB
USB Device: USB 3.0 Bus
USB Device: Intel(R) RealSense(TM) Tracking Camera T265
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

</pre>

</details>

neilyoung, 2020-03-15:
Is there any solution in sight?

radfordi, 2020-03-18:
@neilyoung, I think #5665 (which was merged a few weeks ago) should have fixed this.

neilyoung, 2020-03-18:
Thanks. I must have missed that. But it is not yet available in production, right?

neilyoung, 2020-03-18:
Oh nice... The current development branch does not compile any longer on macOS
Will file a bug

neilyoung, 2020-03-18:
@radfordi  Confirmed. Current development branch (after having fixed #6086 manually as explained) does not have this defect any longer. Thanks. So now the "null pointer" :)

Regards

radfordi, 2020-03-19:
@neilyoung, which issue are you referring to w.r.t the "null pointer"?

neilyoung, 2020-03-19:
@radfordi Oh, sorry, I thought, it would have been obvious. I meant this: https://github.com/IntelRealSense/librealsense/issues/5683

closed 2020-09-21 by RealSenseSupport

neilyoung, 2020-09-22:
Sorry, wasn’t notified about your response.