realsenseai/librealsense #5480: Map another texture to the pointcloud

ohad-denh, 2019-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | { v2.30.0 } | 
| Operating System & Version | { Win 10 } | 

| Platform                                 | PC |

| Language                            |  {python}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Hey everybody,

first avall i wanna say, im new to working with realsense and new to programming in general. Im starting an internship in next year and i have to work with the camera for that. Im using pyrealsense2 in python.

My question is, is it possible to create the
    pointcloud =  rs.pointcloud()
object with an external image as texture? So in the expression 
    pointcloud.map_to(color_frame.get_data())
i want to replace "color_frame.get_data()" with an image, which does not come directly from the camera. Is that possible?

My problem yet is that the map_to function awaits a pyrealsense2.frame type object, so maybe the right question is, can i convert an image or a numpy array to a pyrealsense2.frame-object?

The background is that we take the 2d color_frame.get_data(), edit the data and want to map the edited 2d-image back to the pointcloud.

With best regards,
Oliver

ev-mp, 2019-12-23:
@ohad-denh hello, the prerequisites for a pointcloud using Realsense Depth and 3-rd party RGB streams in the SDK are:
- Finding the intrinsic data of the external camera sensor.
- Calculating the extrinsic matrix between Realsense Depth and the external camera sensor (requires the most effort, in my estimation).
- Using `r2::software_device` to convert the external data format to `rs2::frame` and make the extrinsic/intrinsic binding required for pointcloud generation.
Duplicate of #4202

ohad-denh, 2019-12-23:
Hey,

thank you for your answer. Just to clarify: We don't use a 3rd party RGB stream.

We edit the RGB stream, or the single color_frame, and just want to "re-attach" it to the pointcloud, or calculate the texture of the pointcloud based on the edited version.

We only change the color information. We insert a realsense color_frame into our post-processing, and get an image that reflects 2 types of objects, which are highlighted in red and green, the rest of the image is black.

We want to re-attach the red, green and black picture to the color_frame object or directly to the pointcloud.

Does that change your answer? :)

ev-mp, 2019-12-23:
Thanks for clarification, the result of the custom post-processing is not clear though - do you generate a new image with the colors that highlight the different objects, or modify the original pixels buffer while preserving the `rs::frame` object?

In general we recommend to wrap the user data-manipulation code into processing blocks as this is build to encapsulate both the algorithm and object's state. In addition it allows to preserve the original frame (all the changes are done on deep copies of the frames). For the purpose of prototyping and if the modified RGB frame size is not changed, you can override the frame's internal pixels buffer structure directly (i.e. `frame.get_data()`) with the modified data. It will allow to continue working with the original `rs2::frame` object for pointcloud generation.

In case you select not to use the processing block approach then you need to generate the `software-device` instance, and then populate and use it to convert the customized RGB image back to `rs2::frame` as specified above.

ev-mp, 2020-01-06:
@ohad-denh hello, do you need further support on this topic?
Please update

closed 2020-01-21 by RealSenseCustomerSupport