realsenseai/librealsense #5481: Announcement:  German language YouTube presentation by Laserscanning Europe GmbH about RealSense L515

MartyG-RealSense, 2019-12-18:
Hi everyone,

For German RealSense users, Laserscanning Europe GmbH has released a German-language video that features a discussion about the new RealSense L515 lidar camera.

https://youtu.be/VKTM0on_dHw?t=219

neilyoung, 2019-12-18:
Well, he literally just read from your website https://www.intelrealsense.com/lidar-camera-l515/. No more information contained.

MartyG-RealSense, 2019-12-18:
That is to be expected, given that there is still 4 months until the shipping date.  The main reason for posting it was because people who have English as their second language (or rely on browser auto-translation) appreciate material tuned for their native dialect instead of having to read everything in English.  :)

neilyoung, 2019-12-18:
Of course, you are free to do so :) No problem

closed 2020-01-02 by MartyG-RealSense