realsenseai/librealsense #5482: projection point from depth(rs2::points) to color produced offset

tarekmuallim, 2019-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 0.5.12.01.00 | 
| Operating System & Version |   { Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.31.0}                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

I am working on a program for dimensioning.
Its job is to detect a box(parcel) under the camera and give the dimensions of the box.

My program works good. It process the point cloud of the view and give back the coordinates of the box corner points.

I am trying to draw that box on the color frame coming from the RGB camera.
To do this I calculate the pixel coordinate for each corner point of the box [box_c_x, box_c_y, box_c_z] like that:

```
auto depth_stream = pipe_profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
depth_intrinsics = depth_stream.get_intrinsics();
auto color_stream = pipe_profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
color_intrinsics = color_stream.get_intrinsics();

depth_to_color_extrinsics = depth_stream.get_extrinsics_to(color_stream);
color_to_depth_extrinsics = color_stream.get_extrinsics_to(depth_stream);

float box_c_pixel[2];
float box_c_point[3];

box_c_point[0] = box_c_x;
box_c_point[1] = box_c_y;
box_c_point[2] = box_c_z;

rs2_transform_point_to_point(box_c_point, &sensor.depth_to_color_extrinsics, box_c_point);
rs2_project_point_to_pixel(box_c_pixel, &sensor.color_intrinsics, box_c_point);

```

The results looks correct but with small shift as you see in the image.

[image]

I didn't use rs2::align at all. There is no alignment between the depth and the color frame in my program.

The point cloud is calculated from the depth frame using rs2::points

what is the problem and way I am getting this offset?

ev-mp, 2019-12-18:
@tarekmuallim hello,
The following line seem suspicious as it probably makes API abuse:
>rs2_transform_point_to_point(**box_c_point**, &sensor.depth_to_color_extrinsics, **box_c_point**);

Passing the same object as in and out parameter is not applicable in this scenario as the function requires the input be immutable (check the [implementation](https://github.com/IntelRealSense/librealsense/blob/d19829788008b8e000870895a068f0c43d58895a/include/librealsense2/rsutil.h#L126-L131)) - that's the reason that the third parameter is passed as const ref.

tarekmuallim, 2019-12-19:
Hi @ev-mp I have changed the code and saved the output of rs2_transform_point_to_point to another float array.
But nothing changes I had exactly the same result.

ev-mp, 2019-12-22:
@tarekmuallim, I reviewed the snippet again and there is nothing else I could point to, so you probably need review the other parts of the code, especially the part where the box corners are calculated and extracted.
I would recommend to start with texture mapping validation in `rs-align` example. This will allow to verify the Depth<=>RGB calibration and display the pixels that define the box corners both in Depth and RGB frames. so you'll be able to check them against the values calculated in your code.

tarekmuallim, 2019-12-22:
@ev-mp thanks for your help.
When I apply align color to depth frame I get a perfectly aligned color pixels. I mean the box surface will be filled with right color pixels.
You have to notice that the box corner points coordinates are calculated from the point cloud. Which I assume it is aligned with depth frame coordinate system. Based on that I make the coordinate conversion. 

I also I find that the shift in the calculated pixels coordinates of box corners is fixed whatever the dimensions of the box. That means I get correct shape put all the coordinates are shifted.

ev-mp, 2019-12-22:
So you need to verify that the corners you extract from the pointcloud correspond to the 2D data that you see in the alignment as the key phrase is `the box corner points coordinates are calculated from the point cloud`.
Take the vertices that define the box corners extracted from the point cloud and project them to the Depth frame. Then compare the [i,j] indexes you receive from the projection against the [i,j] indexes you obtain in `rs-align`.
Additionally - can you print the content of `&sensor.depth_to_color_extrinsics` ?

tarekmuallim, 2019-12-23:
Hi
here the calibration information:

> color to depth extrinsics rotation: [ 
>
> 0.999994 -0.000957491 0.00336578 0.000954703 0.999999 0.000829851 -0.00336657 -0.000826633 0.999994 ]
> 
> color to depth extrinsics translation: [ -0.0151473 0.00023473 0.00023473 ]
> 
> depth to color extrinsics rotation: [ 0.999994 0.000954703 -0.00336657 -0.000957491 0.999999 -0.000826633 0.00336578 0.000829851 0.999994 ]
> 
> depth to color extrinsics translation: [ 0.0151465 -0.000285617 -0.000285617 ]
> 
> depth intrinsics model: 4
> 
> depth intrinsics width: 1280
> 
> depth intrinsics height: 720
> 
> depth intrinsics ppx: 618.541
> 
> depth intrinsics ppy: 347.475
> 
> depth intrinsics fx: 974.756
> 
> depth intrinsics fy: 974.756
> 
> depth intrinsics coeffs: [ 0 0 0 0 0 ]
> 
> color intrinsics model: 2
> 
> color intrinsics width: 1280
> 
> color intrinsics height: 720
> 
> color intrinsics ppx: 626.724
> 
> color intrinsics ppy: 365.731
> 
> color intrinsics fx: 924.692
> 
> color intrinsics fy: 923.527
> 
> color intrinsics coeffs: [ 0 0 0 0 0 ]
> 
> depth fov x: 66.5588
> 
> depth fov y: 40.5357
> 
> color fov x: 69.369
> 
> color fov y: 42.5911

I tried to project the box corner points to depth frame by applying:

> rs2_project_point_to_pixel(box_c_pixel, &sensor.depth_intrinsics, box_c_point);

I get exactly the same shift in pixels coordinators which I get for color projection!
The shift is fixed whatever the size of the box.

tarekmuallim, 2019-12-23:
Hi @ev-mp thanks for your help. 
I solved the problem. 
Actually the projection functions in the sdk work fine and have no problem.
In my program, after extraction of the point cloud there was a function which modify the coordinates of point cloud. And this gives the shift in projected coordinates.
Thank again.

closed 2019-12-30 by RealSenseCustomerSupport