realsenseai/librealsense #5485: T265: Q: What is the best strategy to call `get_static_node`?

neilyoung, 2019-12-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

Is it better to issue this request regularly or is it better to wait for a relocalization event? Or in other words: Will an update to the internal pose initially set with `set_static_node` always accompanied by an relocalization event? The latter would be a smarter solution, but I found, that it doesn't work for "freshly set" static_nodes. Means: I get a relocalization event, when I'm starting to track on an existing or newly loaded map, but I don't get such, if I call `set_static_node.

So a mix of both appropriate?

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi @neilyoung

With the samples that are out there do you still need anything on this thread?

The rs-ar-advanced (https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-advanced) is out there in librealsense.

neilyoung, 2020-02-03:
@RealSenseCustomerSupport Old issue. Can be closed.

closed 2020-02-03 by neilyoung