realsenseai/librealsense #5486: How to calculate the coordinate offset and angle between two frames?

Dontla, 2019-12-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to make the car realize autonomous driving on unstructured roads. During the process of the car, use D435 installed vertically to capture the frames. Is there a way to calculate the deflection angle between the two frames, or the deflection of the coordinates? The amount of movement so that we can know how far the car has traveled over a period of time and how much the angle has changed. Do you have such a similar code example or pyrealsense function?

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@Dontla Please refer to the example below(SLAM with D435i) which based on RTAP-Map.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
http://introlab.github.io/rtabmap/

Also you can refer to https://github.com/tzutalin/awesome-visual-slam#projects for more projects.

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@Dontla Did you get any chance to try? Any other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@Dontla If no other questions about this, will close it accordingly. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport