realsenseai/librealsense #5487: Two Depth Units parameters in viewer

Jiloc, 2019-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 

I would like to reduce the depth units to avoid the quantization effect at low range. I am working at very close range, 11-13 cm.
I already achieved decent results adjusting the depth units and the disparity shift, but I'd like to understand why there are 2 different depth units parameters in the viewer.

One is in the general Controls menu
[image]

The other one is under Advanced Controls -> Depth Table
[image]

Which one should I look into? How are they related? What is the lowest (working) value for the depth units that you tested?

dorodnic, 2019-12-22:
These two sliders are mapped into the same underlying register. The reason for redundancy is that all depth capable devices offer `RS2_OPTION_DEPTH_UNITS` (usually as read-only option, to allow projection functionality). In addition, D400 allows setting new value via Depth Table controls (prior to streaming if I recall correctly). 
I'm not sure regarding lowest working value, perhaps @agrunnet could comment on that

RealSenseCustomerSupport, 2019-12-31:
----------------------------------------------

Hi @Jiloc Generally, the lowest value of depth unit is 100.
Thanks,

Jiloc, 2020-01-03:
Thank you both for the answers.

@RealSenseCustomerSupport Sorry but I am still a little bit confused. Since both of the depth units variables are connected to the same register, both values should be important. Do you mean using a Controls -> Depth Units = 0.001 and a Advanced Controls -> Depth Table -> Depth Units = 100?
 Which corresponds to 10μm? 
@dorodnic after a long debug I found out that setting RS2_OPTION_DEPTH_UNITS to a sensor of a  `rs2::software_device` doesn't work with the `rs2::align` filter. It is always 0. During the debugging I realized that during the align processing, instead of getting the RS2_OPTION_DEPTH_UNITS value, the device would read the RS2_OPTION_STEREO_BASELINE. Setting the RS2_OPTION_STEREO_BASELINE with the depth units value will make it work.

If I'll have more time I'll try to give additional information but the

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

I'll try to answer your questions:
The 2 depth unit controls are connected but I recommend changing the one under Controls (not in Advanced Controls). Doing this should automatically change the other to same or equivalent value and will display the properly scaled depth. If you change in Advanced controls, you will need to stop and start streaming to get the correct depth values displayed and the value under Controls may not get updated correctly to the new value.

The units are meters in the controls section and micrometers in the Advanced section. So .001 under controls is the same as 1000 in Advanced.

With the latest version of FW (5.12.01 or later), increments of 1 micron should be available. But, keep in mind that the MaxZ will be reduced significantly for such a small step size (to ~65mm) and you would need a very large disparity shift. This is really intended for smaller BL cameras.

I would recommend using 10 microns (0.00001 under Controls and will show up as 10 in Advanced). It will be truncated in display as .0000 since it's currently limited to 4 significant digits. This will give you a MaxZ of ~65cm.

Let me know if this helps.

Jiloc, 2020-01-08:
crystal clear, thank you

closed 2020-01-08 by Jiloc

Dilip-Gurumurthy, 2020-07-21:
> I'll try to answer your questions:
> The 2 depth unit controls are connected but I recommend changing the one under Controls (not in Advanced Controls). Doing this should automatically change the [...]

I am unable to set it to 10 microns(0.00001). The following error is shown,
"1e-05 is out of bounds [0.0001,0.01]

My FW version is: 05.11.01.100
Model: D415