realsenseai/librealsense #5490: Resource not found: realsense2_camera

kremerf, 2019-12-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   {Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (5.0.0.37)                                         | 
| Platform                                 | NUCi7  |
| SDK Version                            |  { legacy / 2.31.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot/Camera }                          | 

### Issue Description
Hi,

i am currently using a D435 and just recently re-installed Ubuntu, ROS and the RealSense SDK to get a fresh platform. Unfortunately I cannot use the "realsense_viewer" via SSH from my Computer.

It starts, but i get following error message inside the realsenseviewer:

`Backend in rs2_open_multiple(sensor:0x7fa288001740, profiles:0x7fa2881427a0, count:1): xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy`

In my terminal it sais:

 ```
19/12 11:52:51,110 WARNING [139842113496832] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 19/12 11:52:51,118 ERROR [139842113496832] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
```

On the NUC itself, the software runs just fine, i can see depth, RGB and both together in 60 FPS.

Additionally and maybe related: When i start my launch-file to start up my project and the camera together, i get the following error message:

```
Resource not found: realsense2_camera
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/paytah/catkin_ws/build/catkin_tools_prebuild
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file
```

ltorquato4, 2019-12-19:
Hi @kremerf ,
I am having the same problem since I upgrade my SDK version. Did you test it already through the command : realsense-viewer ?

kremerf, 2019-12-19:
Yes, the program (realsense-viewer) itself loads on the NUC and on the via SSH-Connected computer. But just the NUC is able to retrieve the visual data.

kremerf, 2019-12-19:
I updated the realsense firmware and now I do get an image on both devices. The last error "Resource not found: realsense2_camera" remains though.

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

kremerf, 2020-04-30:
Hey,

No I think I solved it, but I cannot remember exactly how, as this topic is somewhat older..
But I often noticed that deleting build/devel/install spaces in Catkin solved issues for me, after updating my system. As rebuilding them fixes a lot of broken connections.

closed 2020-04-30 by kremerf

YoanDev, 2022-03-01:
Hi, I had the same issue. I solved it with the command "sudo apt-get install ros-$ROS_DISTRO-realsense2-camera"

tharun-selvam, 2024-05-09:
@YoanDev I tried the above command and it seems the package is already installed. I am still getting the 'resource not found' error