realsenseai/librealsense #5491: New version of SDK doesn't find camera

ltorquato4, 2019-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | ? | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-72-generic | 
| Platform                                 | PC |
| SDK Version                            |  legacy / 2.31.0 | 
| Language                            |  python | 
| Segment			|  Robot  | 

### Issue Description
Hi, 
I just installed the newest version of Intel RealSense SDK (2.31.0) but it doesn't recognize my camera anymore. I tried already to find already a way to downgrade it, but it seems I have to install everything again. Anybody knows an easier way to downgrade SDK or can explain why this is happening? 
Thank you :)

Debitsch, 2019-12-19:
Same for me on a Windows 10 system. After a downgrade  from 2.31 to 2.29 the D435 works again.

ev-mp, 2019-12-20:
@ltorquato4 , do you have kernel/application logs for review?

ltorquato4, 2019-12-20:
Hi @ev-mp ,
I used in "realsense-viewer" in the command line in order to launch the Intel RealSense Viewer, it opened but it wasn't able to recognize the camera. 
Also used the pyrealsense2 library, with a code that was working for the last release of IntelRealSense SDK, but it didn't work. 
Now I reinstalled everything again, but still it doesn't work.

ltorquato4, 2019-12-20:
Hi @Debitsch ,
Could you explain me how you did your downgrade to this version?

Debitsch, 2019-12-20:
I compiled the library using the source Code from GitHub. Using the latest Version (2.31.0) it was not possible to connect to the D435. Using the Version tagged with 2.29.0 worked.

Note: I tried it with a T265 too. For this sensor both Versions (29 + 31) worked.
Note: For Building I used the Visual Studio 2017 c++ Compiler.

ev-mp, 2019-12-23:
@Debitsch , did you try to use the prebuild `realsense-viewer` for v2.31.0 from the [release page](https://github.com/IntelRealSense/librealsense/releases/download/v2.31.0/Intel.RealSense.Viewer.exe) ?

@ltorquato4 , please enable SDK logs in `realsense-viewer` under the `settings` in the upper right corner:
[image]
Then re-plug the camera and share the log for review.

Debitsch, 2019-12-23:
The release build you provided works.

My Cmake configuration (in PreLoad.cmake) is
```
set(BUILD_EASYLOGGINGPP         ON              CACHE BOOL "override" FORCE)
set(CENTRAL_BL_VERSION          ON              CACHE BOOL "override" FORCE)
set(CMAKE_CONFIGURATION_TYPES   Debug;Release   CACHE STRING "override" FORCE)
set(CMAKE_DEBUG_POSTFIX         ""              CACHE STRING "override" FORCE)
set(BUILD_WITH_TM2              ON              CACHE BOOL "override" FORCE)
set(BUILD_SHARED_LIBS           OFF             CACHE BOOL "override" FORCE)
set(ENABLE_CCACHE               OFF             CACHE BOOL "override" FORCE)
set(BUILD_EXAMPLES              ON              CACHE BOOL "override" FORCE)
set(BUILD_WITH_STATIC_CRT       OFF             CACHE BOOL "override" FORCE)
set(FORCE_RSUSB_BACKEND         ON              CACHE BOOL "override" FORCE)

set(CMAKE_INSTALL_PREFIX ${CMAKE_CURRENT_BINARY_DIR} CACHE PATH "override" FORCE)

set(CMAKE_SYSTEM_VERSION 10.0.16299.0 CACHE TYPE INTERNAL FORCE)
set (CMAKE_GENERATOR "Visual Studio 15 2017 Win64" CACHE INTERNAL "" FORCE)
```
using the build commands in a Visual Studio command prompt:

```
if exist Build rmdir /S /Q Build
mkdir Build
cd Build

cmake ..\

msbuild Build\INSTALL.vcxproj /p:Configuration=Release
msbuild Build\INSTALL.vcxproj /p:Configuration=Debug

```

ev-mp, 2019-12-23:
@Debitsch , the Realsense devices are not recognized as you have the  `FORCE_RSUSB_BACKEND` option turned on
>set(FORCE_RSUSB_BACKEND         ON              CACHE BOOL "override" FORCE)

The flag tells the SDK to replace the standard MS UVC driver with a different backend that also requires installation of Realsense INF files (Redistributed with [Win7 Installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.31.0/Intel.RealSense.SDK-WIN7-2.31.0.1234.exe)). The main use-case of this alternative backend is with Windows7 that doesn't provide a viable UVC support.

You need to rerun the CMake with this flag disabled, and then rebuild the solution 
>cmake ..\ -DFORCE_RSUSB_BACKEND=false

Debitsch, 2019-12-23:
After using the correct configuration for the FORCE_RSUSB_BACKEND flag, Version 2.31 works fine.

Time to read the docs …

Thanks,
Rasmus

RealSenseCustomerSupport, 2019-12-30:
----------------------------------------------

Hi @ltorquato4, Does your issue resolved as well?
Thanks,

closed 2020-01-23 by RealSenseCustomerSupport

ltorquato4, 2020-01-24:
Hi,
@RealSenseCustomerSupport sorry for not answering in time, just to let you know that it worked with 2.29.0 version, but with the new release it didn't. 
Also with the new release I wasn't able to obtain the SDK logs in realsense-viewer.
I will keep using the old version of SDK.
Thank you for your attention,
Best regards.

ltorquato4, 2020-01-24:
By the way, I just saw the new release [2.32.1](https://github.com/IntelRealSense/librealsense/issues/5704) . 
I will try this out!