realsenseai/librealsense #5493: RGB Exposure custom value

amatabuena, 2019-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  --                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  2.19.0                          | 
| Language                            |  C#                         | 
| Segment			|  others                          | 

### Issue Description
Due to better image capturing in specific conditions, I need to configure Color Exposure in manual mode. When doing it, and although Option Exposure sais that Step value is 1, I can't set the specific value I need.

For example, both with Intel.Realsense.Viewer and directly with C# wrapper, when I try to put color exposure value to 600, 610 or 650..., internally the sensor sets the exposure to 625. This behavior implies that exposure can't be adjusted finaly. 

For clarifing the main reason to don't using AutoExposure is that capturing a face located in a near range (60 - 80 cm) with a quite high light on the back causes the image to be over exposed. It would be possible to set the minimun and maximun exposure allowed with autoexposure setted on?

ev-mp, 2019-12-20:
@amatabuena, this is a limitation of the Microsoft MediaFoundation driver uses logarithmic scale to control the sensor's exposure value:
https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/ksproperty-cameracontrol-exposure.
While on Linux you can control the exposure with 0.1 msec granularity. On Windows with the standard UVC driver you're limited to **manually select**\* between probably 5 or 6 discrete values. And attempts to set off-scale values will be "rectified" by MediaFoundation infrastructure.
*When Auto-Exposure is set on then the sensor controls the exposure internally using the actual control's range and not the one imposed by OS.

Just to explain with given example - the value that you read is 625. If you inspect the Librealsense's exposure control description then you'd find that it reports 0.1msec units, i.e  625=>>62.5 msec, and 62.5 msec corresponds to 1/16 of a seconds, or  (-4) using MS logarithmic scale.
If you try to use 600/610/650 values the MF rounds it up to 625 - the nearest value on the logarithmic scale.

amatabuena, 2019-12-20:
Hi @ev-mp ,

Thansk for the answer. Asking through my ignorance, It would be possible to use on Windows a driver different of Microsoft Media Foundation?

ev-mp, 2019-12-23:
@amatabuena , the SDK provides an alternative to the standard MS driver/API with `RSUSB` infrastructure that bypasses the MS UVC driver and implements the UVC protocol in user-space. See more in the [Release Notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements).
To select this option you need to activate the `FORCE_RSUSB_BACKEND` Cmake flag.
Note, however, that using the alternative driver also requires to install custom camera drivers.

We provide a precompiled SDK installer that is built with the alternative driver and that you can try it out of the box - 
https://github.com/IntelRealSense/librealsense/releases/download/v2.31.0/Intel.RealSense.SDK-WIN7-2.31.0.1234.exe
Just remember that you mark the `install drivers` checkbox during the installation wizard and to remove those during uninstall operation in order to use the standard MS UVC infrastructure.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@amatabuena Did you get it through with ev-mp's reply? Looking forward to your update. Thanks!

closed 2020-01-10 by RealSenseCustomerSupport