realsenseai/librealsense #5496: Issue Calibrating D435i using python calibration script

MaxxWilson, 2019-12-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.01 | 
| Operating System & Version  | Ubuntu 18.04 | 
| Kernel Version (Linux Only)   | 5.0.0-37-generic | 
| Platform                                  | PC |
| SDK Version                           | 2.31.0 | 
| Language                               | Python/C++ | 

Hello,

I am trying to calibrate my new D435i using the rs-imu-calibration.py script. The camera detects motion even when perfectly still and continually posts "WARNING: MOVING." I ran rs-motion and the camera model was spinning erratically although the camera was still. I saw a similar issue posted yesterday but it was closed without a response.

lishanggui, 2019-12-20:
I have the same problem.when I print the gyroscope values from  rs-imu-calibration.py, I found the  gyroscope values  is  not about [0,0,0].  the values is about [ 6.98131695e-03 5.67720680e+01 -2.15338726e+01]

ev-mp, 2019-12-23:
@MaxxWilson  hello, please use the [instructions in #5515](https://github.com/IntelRealSense/librealsense/issues/5515#issuecomment-568374098) to collect raw IMU data for review.
Two more things to clarify:
- Do you use Debian distribution or build the SDK manually?
- What is the current camera calibration ? - `rs-enumerate-devices -c`

MaxxWilson, 2019-12-23:
I used the Debian Distributions.

Here is the raw IMU data:
[data_log.txt](https://github.com/IntelRealSense/librealsense/files/3994393/data_log.txt)

Here is the current camera calibration:
[rs-enumerate_output.txt](https://github.com/IntelRealSense/librealsense/files/3994401/rs-enumerate_output.txt)

ev-mp, 2019-12-23:
@MaxxWilson thank you.
We will review the data

radfordi, 2020-01-09:
@MaxxWilson, this looks looks to same regression reported in #5515 where the [suggestion](https://github.com/IntelRealSense/librealsense/issues/5515#issuecomment-568758744) was to use an older version.

RealSenseCustomerSupport, 2020-04-30:
----------------------------------------------

Hi,

Are you still experiencing issues with this. The regression appears to have been fixed as of Ver 2.32.1 of librealsense.

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-08 by RealSenseCustomerSupport