realsenseai/librealsense #5498: imu calibration problem about rs-imu-calibration.py

lishanggui, 2019-12-20:
When I run the program python3 rs-imu-calibration.py in the first position according to the instructions, the program keeps repeating WARNING: MOVING and Status.wait_to_stable, but cannot perform calibration.as follow,

cti@cti-desktop:~/librealsense/tools/rs-imu-calibration$ python3 rs-imu-calibration.py 
Start interactive mode:
FOUND GYRO with fps=200
FOUND ACCEL with fps=63
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Upright facing out
   Status.rotate:           [-0.0031 -0.0023 -0.0681]:                 [ True  T                                                                                 Status.collect_dataWARNING: MOVING
   Status.rotate:           [-0.0052 -0.0021 -0.064 ]:                 [ True  T                                                                                 Status.collect_dataWARNING: MOVING
   Status.rotate:           [-0.0072 -0.0021 -0.064 ]:                 [ True  T                                                                                 Status.collect_dataWARNING: MOVING
   Status.rotate:           [-0.0052 -0.0021 -0.0641]:                 [ True  T                                                                                 Status.collect_dataWARNING: MOVING
   Status.rotate:           [-0.0052 -0.0019 -0.0619]:                 [ True  T

 When printing the gyroscope values in the program, I found the gyroscope The value of the meter has always been kept large, and the gyroscope data cannot be collected, which makes it impossible to calibrate. How to solve this problem?

collect  gyroscope values code as follow:
                    if pr.stream_type() == rs.stream.gyro:
                        self.collected_data_gyro.append(np.append(frame.get_timestamp(), data_np))
                        is_moving = any(abs(data_np) > self.rotating_threshold)

gyroscope values is about [ 6.98131695e-03  5.67720680e+01 -2.15338726e+01]

ev-mp, 2019-12-23:
@lishanggui hello, the symptoms you describe resemble the #5515 so please more details to understand whether there is a connection:

1. Which platform are you running, is it Arm or X86 ?  
2. Are you using Debian distribution or build the SDK from Source? For manual  build - which CMake flags do you configure and what is the value of `FORCE-RSUSB-BACKEND` ?  
3. What are the SDK and FW versions that you use ? Use `rs-enumerate-devices -c` to gather the device info.
4. Please also generate and provide data for analysis using the supplied rs-data-collect tool:
    - Create config file data_collect.cfg that contains the following two lines :
    **ACCEL,1,1,63,MOTION_XYZ32F
    GYRO,1,1,200,MOTION_XYZ32F**
    - Run the collection tool when the camera is idle with the command line parameters:
    ```rs-data-collect -c ./data_collect.cfg -f ./data_log.csv -t 20```

lishanggui, 2019-12-27:
@ev-mp thank you!  I have slove the problem.

closed 2019-12-30 by RealSenseCustomerSupport

shoaibazam, 2020-01-11:
@lishanggui @ev-mp  Hi, I have encountered this problem Status.collect_dataWARNING: MOVING while calibrating the d435i camera. Please help me in this regard.
Thanks.

lishanggui, 2020-01-11:
@shoaibazam  Please try to uninstall the software and try to install with librealsense source code. Do not use pip install pyrealsense2 to install python programs.