realsenseai/librealsense #5499: How to get the intrinsics with numpy in Python?

JimXu1989, 2019-12-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<30>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I want to transform the cpp code:  https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_types.h
 to python.
Which need to get the intrinsics to calculate. 

Here is my code:
import pyrealsense2 as rs2
import numpy as np
a = np.asanyarray(rs2.intrinsics.height)
a
the put put is as follow:
array(<property object at 0x00000154A07EECC8>, dtype=object)

How could I get the intrinsics in real number?
Thanks a lot!

neilyoung, 2019-12-20:
Check out the t256_stereo.py sample in wrappers/python/examples

JimXu1989, 2019-12-20:
> Check out the t256_stereo.py sample in wrappers/python/examples

Oh, it seems if I want to get the intrinsics, I should plug the sensor first. Am I right?

neilyoung, 2019-12-20:
I think so

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @ JimXu1989
Do you need further help on this?
Thanks,

closed 2020-01-15 by RealSenseCustomerSupport