realsenseai/librealsense #5500: How to get the height images in frames using depth/rgb images. ——libRealsense1, ZR300

NooBlock, 2019-12-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { ZR300 } | 
| Operating System & Version |    Linux (Ubuntu16 | 
| Kernel Version (Linux Only)    |  (4.4.0-170) | 
| Platform                                 | PC|
| SDK Version                            |  { legacy v1.12.1 + realsense_sdk_zr300 }| 
| Language                            |  {c++ opencv }                          | 
| Segment			|  {others }                          | 

### Issue Description
Since the librealsense-v1 legacy stops updating at the v1.12.1, and now I'm using a ZR300 in order to get real-world coordinates. But the main task is to get the height of every point in the depth or rgb map.For example,  I want to output a frame of the points where the value at point x, y shows the real-world(or relative height to camera ) height instead of the depth.
The fucntion of "rs_deproject_pixel_to_point" at rsutil.h seems to work but actually it doesn't do anything of showing the z-axis's value.
 Any idea to get the height of the point at the map?（It's OK to get the relative height to the camera. ）

Thanks.
Block

dorodnic, 2019-12-20:
Hi @NooBlock 
Relative height from the camera would be the Y coordinate produces from deproject function. 
Also, for your use case you may find this code useful:
[depth-metrics.h:62-111](https://github.com/IntelRealSense/librealsense/blob/master/tools/depth-quality/depth-metrics.h#L62-L111)
It is estimating 3D plane from a collection of 3D points, and can be used as basis of floor plane detection (run it several times and every time drop the outliers and you should converge to the ground plane)
Here is some extra documentation on deprojection - [Projection-in-RealSense-SDK-2.0](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0)

NooBlock, 2019-12-23:
> Hi @NooBlock
> Relative height from the camera would be the Y coordinate produced by the deproject function.
> Also, for your use case you may find this code useful:
> [depth-metrics.h:62-111](https://github.com/IntelRealSense/librealsense/blob/master/tools/depth-quality/depth-metrics.h#L62-L111)
> It is estimating 3D plane from a collection of 3D points, and can be used as basis of floor plane detection (run it several times and every time drop the outliers and you should converge to the ground plane)
> Here is some extra documentation on deprojection - [Projection-in-RealSense-SDK-2.0](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0)
@dorodnic 
Thank you very much! I'll try it first.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@NooBlock Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport