realsenseai/librealsense #5502: RGB Image from python realsense is very dark

arifsaeed, 2019-12-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer 2.31.0--> Click info) | 
| Operating System & Version |   {Win (10)| 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { pyrealsense2-2.31.0}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
I would like to get the RGB image using python,and wrote the code below to extract the image. however when i display the image it is very dark, however when i use the realsense viewer, the picture is not dark at all.  Can you please advise how i can get the same quality image from python that I can get from the realsense viewer

below is the code i use with the two pictures:

import pyrealsense2 as rs
import numpy as np
from open3d import *     
from generatepointmap import generate_pointcloud
import cv2
import pdb
#import pptk

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

cfg=pipeline.start(config)
profile = cfg.get_stream(rs.stream.depth) # Fetch stream profile for depth stream
intrinsic = profile.as_video_stream_profile().get_intrinsics()
ph_intrinsic=camera.PinholeCameraIntrinsic()
ph_intrinsic.set_intrinsics(intrinsic.width, intrinsic.height, intrinsic.fx, intrinsic.fy, intrinsic.ppx, intrinsic.ppy)
depth_scale=cfg.get_device().first_depth_sensor().get_depth_scale()
#pdb.set_trace()
while True:
    
    frames = pipeline.wait_for_frames()
    depth_frame = frames.get_depth_frame()
    color_frame = frames.get_color_frame() 
    align_to = rs.stream.depth
    align = rs.align(align_to)
    aligned_frames = align.process(frames)
    color_array=np.array(color_frame.get_data())#set_color(np.array(color_frame.get_data()))
    depth_array=np.array(depth_frame.get_data())
[image]
[image]

    cv2.imshow("image", color_array)
    cv2.waitKey(0)

arifsaeed, 2019-12-21:
i have been playing with the camera today, and now i can see some strange behaviour.

If i run my code consecutive times without restarting my python console. then the first picture is dark as befor, but the second picture becomes brighter, but the color of the arm has changed to yellow, which is not correct.  The arm is actually blue
See three pictures below

[image]
[image]

dorodnic, 2019-12-21:
Hi @arifsaeed 
Please note that opencv default is actually BGR, not RGB, so I believe it has something to do with the artifact. 
You can configure the camera to bgr8 instead of rgb8.
As for the dark image, this may have something to do with auto-exposure convergence time. Does the image continue to be dark if you take another ~100 frames or does it go back to normal?

RealSenseCustomerSupport, 2019-12-30:
----------------------------------------------

@arifsaeed Do you get it through with the reply from dorodnic? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@arifsaeed Didn't get response from you for weeks. Will close it at this point. Please feel free to create new ticket if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport

knowlessthanenough, 2024-01-24:
i also have the same problem, using viewer is normal however using python to call it return very dark RGB image

knowlessthanenough, 2024-01-26:
thanks a lot, i have fix the problem

MartyG-RealSense, 2024-01-26:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!