realsenseai/librealsense #5505: about Modified resolution

ltz17, 2019-12-21:
using config.enable_stream(rs.stream.depth,width, height, rs.format.z16, frame_rate)
Is width=640， height=360 depth as accurate as  width=1280， hieght=720 depth?
Is it just downsampling or is there a change in speckle and accuracy?
how can I get a good accurate at 640*360 depth?

MartyG-RealSense, 2019-12-21:
The advantages of using a low depth resolution is that less data is generated (less USB bandwidth, and smaller file sizes when recording data).  But depth precision will be degraded at lower resolutions 
 Intel's camera tuning guide states that the optimal resolutions are 1280x720 for D415 and 848x480 for D435.

You can view the guide here:

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

Assuming that using a higher resolution is not an option for you, the camera tuning guide also describes the advantages of using a 'High Accuracy' camera configuration "preset" to help achieve greater accuracy.  

To quote the guide:

************

The “High Accuracy” preset will impose stricter criteria, and will only provide those depth values that are generated with very high confidence.  This will give a less dense depth map in general but is very good for autonomous robots where false depth, aka. Hallucinations, are much worse than no depth. 

To illustrate this point, imagine that you are making a robot that needs to walk across lava. The robot will need to know with 100% accuracy that it is stepping on protruding solid rock, or the effects will be disastrous. 

Of course the extreme case is that we are too conservative on depth, and the robot does not see a real obstacle that is in front of it, but the recommended settings should be good.

For more information on Depth Presets, and pictures for each, please refer to this:

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

*************

The link below has code for setting the High Accuracy preset using scripting.

https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Closing this ticket if no further question on the same topic.
Thank you.

closed 2020-01-10 by RealSenseCustomerSupport