realsenseai/librealsense #5506: Angle on pointcloud is not accurate

XuanLin, 2019-12-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   {Ubuntu 16} | 
| Kernel Version (Linux Only)    |  (4.15.0-65-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {pyrealsense 2.30.0.1216}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

Hi,

I am trying to use realsense d435 to reconstruct a point cloud of a paper box and measure the angles between each face pair. All the angles are 90deg angles, but I am getting results that are larger (about 110 deg). For example, I take a picture:
[image]
use masking technique to segment only the depth info. associated with the box out, and use deproject function in librealsense to reconstruct the point cloud, I get a result that looks like:
[image]
Obviously, the box angle that the point cloud shows is larger than 90 deg, and the plane-fitting result shows 120deg. I also tried to take pictures from another angle:
[image]
and the point cloud still shows more than 90 deg angle (in this case it is 110deg):
[image]

I try to implement the temporal filter in realsense library, and it doesn't help. Does anyone know why there is such a big error? It looks like a very consistend distortion, since all the right angles are measured larger than 90 deg.

The box is about 10cmx10cmx10cm large, made by thick white paper, with some random pattern.

Thank you!

dorodnic, 2019-12-21:
Hi @XuanLin 
This is very strange. I would recommend capturing the same scene using the RealSense Viewer and exporting pointcloud to PLY (from the 3D view), just to eliminate any issues with the algorithm. 
What is the distance to the box?

XuanLin, 2019-12-22:
> Hi @XuanLin
> This is very strange. I would recommend capturing the same scene using the RealSense Viewer and exporting pointcloud to PLY (from the 3D view), just to eliminate any issues with the algorithm.
> What is the distance to the box?

The distance is about 10 cm, I just try to fill the view with the box.

I look at the 3D plot of realsense-viewer, it looks like the point cloud should be 90 degrees. I think there should be some bugs in my point cloud processing code...

Additionally, I have to put the camera with 20-30 cm distance from the box in the realsense viewer, otherwise the point cloud is blocked out. I try to place the camera further away (like below), but my code still gives something between 100 to 110 degrees.

[image]

XuanLin, 2019-12-22:
I suddenly realize what may be wrong. In order to do segmentation I align depth image to RGB image. But the depth and RGB camera has different FOV. So this alignment/rescale changes the focal length ...

closed 2019-12-22 by XuanLin

maky-hnou, 2019-12-28:
@XuanLin  is there a built-in method in the SDK that measures angles? 
I only know how to measure distance between the camera and objects.

XuanLin, 2020-01-01:
> @XuanLin @dorodnic is there a built-in method in the SDK that measures angles?
> I only know how to measure distance between the camera and objects.

I was not aware of that. I implement my own plane fitting method to measure angles.

maky-hnou, 2020-01-04:
Is it possible to describe your method?

XuanLin, 2020-01-08:
> Is it possible to describe your method?

Right now I just use pretty straightforward method. Basically, I implemented semantic segmentation in RGB image (MaskR-CNN) that segment the object out, apply the same mask on point cloud data, and use RANSAC to do plane fitting on point cloud.

maky-hnou, 2020-01-08:
@XuanLin thank you