realsenseai/librealsense #5507: Depth map edge depth value is 0

So-LiYatao, 2019-12-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.06.250 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.1                        | 
| Language                            |  C++                 | 

### Issue Description
The depth map shows that the edge depth value of the object is 0. Especially when small targets are encountered, the depth map depth value is missing, resulting in insufficient point cloud data. I want to know what caused the missing edges and how I can compensate
[image]
[image]
[image]
[image]
[image]

dorodnic, 2019-12-21:
Hi @So-LiYatao 
There are many reasons for zero depth, but the main core issue is the effect of occlusion. Whenever you have two distinct view ports in 3D, you are going to find points in one view that are occluded in the other. In ideal scene with only flat planes perpendicular to the camera, you will get zero occlusion and hence very clean edges, but if geometry has a lot of convex parts (say a ball), points closer to the camera will inevitably occlude points near the edge. 
In D400 products there are 3 different types of occlusion artifacts - 
a. Occlusion between left viewport and right viewport (main cause of black outlines)
b. Occlusion between left viewport and the projector (reduces the amount of details near edges)
c. Occlusion between left viewport and the RGB sensor (causing artifacts near edges in texture mapping) 

We provide simple heuristics to combat occlusion, check out `Post Processing` block in the Viewer and the relevant code-samples. More complex solution could involve two cameras to get information from occluded areas.

RealSenseCustomerSupport, 2019-12-27:
----------------------------------------------

Hi @So-LiYatao, Do you need further help on this?
Thanks,

closed 2020-01-28 by RealSenseCustomerSupport