realsenseai/librealsense #5508: run buildlibrealsense.sh get the "version mismatch!" error

Andrew5c, 2019-12-21:
I run the installlibrealsense.sh first,and run the buildlibrealsense.sh .but the result is "version mismatch!",what is the problem?
it will be very thanks for a reply.

dorodnic, 2019-12-21:
Hi @Andrew5c 
What are `installlibrealsense.sh` and `buildlibrealsense.sh`? Can you provide more information please?
Alternatively, please review official installation guides under [`/doc`](https://github.com/IntelRealSense/librealsense/tree/master/doc) and let us know if you encounter any problems

Andrew5c, 2019-12-21:
Hi, thanks for your timely reply first. @dorodnic ,I install the librealsense use this script:[https://github.com/JetsonHacksNano/installLibrealsense], but when I run the buildlibrealsense.sh in my NVIDIA nano, the terminal notice me "version mismatch!". I don't know what to do. have you meet this problem before?
thanks again.

dorodnic, 2019-12-21:
Got it. 
Just to clarify - JatsonHacks is an independent site, not directly affiliated with Intel RealSense. Jim is super friendly answering questions. We also have an up-to-date official guide for installation on the NVidia Jetson: [`doc/installation_jetson.md`](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) which is essentially equivalent to `installlibrealsense.sh`.
Please note that `buildlibrealsense.sh` and `installlibrealsense.sh` are two alternative methods to get librealsense, and you don't have to run both. If `buildlibrealsense.sh` was completed successfully you now have librealsense v2.31 installed. `installlibrealsense.sh` is trying to install v2.30.1 I believe, hence the mismatch. You can remove any remains of v2.30.1 using `sudo apt remove librealsense2`.

dorodnic, 2019-12-25:
@Andrew5c 
Any luck with the installation?

Andrew5c, 2019-12-26:
@dorodnic ，Thank you very much for your careful help first. I haven't try your method. because I have a lot of classes to take these days. I will try the method that you told me tomorrow. and then I will give you a feedback. Thanks again.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Hi Andrew. Have you tried dorodnic's suggestions? Will you be needing further assistance with this?

closed 2020-01-13 by RealSenseCustomerSupport

Andrew5c, 2020-04-14:
@dorodnic ,Hi, I reinstall the librealsense recently. but I met some problems, hope you can give me some advices.
1. I install the librealsense successfully follow the guides of Jetsonhacks, by run the `./installlibrealsense.sh`, and I run the `realsense-viewer` successfully. but the problem is that I don't see the SDK.
2. After that, I run the `sudo apt remove librealsense2` to remove the above install for next build. And I success to remove it.
3. I run the `./buildlibrealsense.sh` in Jetsonhacks git hub: [https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh](https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh)
. and again successfully builded. But the problem is **when I run the realsense-viewer, and plug the T265 in Nano, it show nothing! no picture, no device. not like the case in STEP 1.** 
can you give me a hand . Thanks very much!!!