realsenseai/librealsense #5509: T265 crashes when used with D435 (Raspberry Pi, Ubuntu, Python)

Leijoma, 2019-12-22:
Hi, 

I am (trying to) use a the T265 and a D435 for a wheeled robotic project. 

The code is written in Python (ver 2.7.15+) running on a Raspberry Pi 3+.

When i run the cameras using Real Sense Viewer al lis good. 

When i run the code with one camera it works as expected.

But when i try to initialize both cameras at the same time the T265 crashes after a few seconds with this error message:

14:49:14.805 [1708] [E] Device-91A0: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
14:49:15.012 [1697] [E] Device-91A0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
14:49:15.013 [1697] [E] Device-91A0: Entered state [ERROR_STATE]
Traceback (most recent call last):
File "camtest.py", line 30, in <module>
frames = pipeline_1.wait_for_frames()
RuntimeError: Frame didn't arrived within 5000  

If i comment out the code related to one of the cameras the code works just fine, but with both cameras enabled the T265 code crash.

This is the code i run:
`
import pyrealsense2 as rs

#265
pipeline_1 = rs.pipeline()
config_1 = rs.config()
#config_1.enable_device('909212110600') #T265
config_1.enable_stream(rs.stream.pose)
pipeline_1.start(config_1)

# D435
pipeline_2 = rs.pipeline()
config_2 = rs.config()
config_2.enable_device('831612070347') #D435
config_2.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
pipeline_2.start(config_2)

print('started')
while True:
   frames = pipeline_1.wait_for_frames()
   pose = frames.get_pose_frame()
   if pose:
      data = pose.get_pose_data()
      pos = {'x': data.translation.x, 'y': data.translation.y, 'z': data.translation.z,'xr': data.rotation.x, 'yr':     data.rotation.y, 'zr': data.rotation.z, 'wr': data.rotation.w}
      print(pos['x'])
 
   frames2 = pipeline_2.wait_for_frames()
   depth = frames2.get_depth_frame()
   if depth:
      dist = depth.get_distance(100, 250)
      print('distance: ',dist)
`
Any help appreciated!

MartyG-RealSense, 2019-12-22:
This question has been given a reply on the Intel Support forum's copy of it.   Thanks!

https://support.intelrealsense.com/hc/en-us/community/posts/360038053533/comments/360009991093

radfordi, 2020-01-08:
@Leijoma, this is working for me since #5213 was merged.  This should work when v2.32.0 is released.

zwn, 2020-02-08:
@radfordi I have the opposite experience. See my comment here: https://github.com/IntelRealSense/librealsense/issues/3434#issuecomment-583477888. It is working with 2.31 and not working with 2.32. See also  #5614.

minihops123, 2020-02-18:
Hi,
I played now several days with the T265+ D435i + Raspberry Pi 4b + Raspbian buster arround. I tried several versions of librealsense (2.32.1, 2.31, 2.29) and all have issues. 
The latest (2.32.1) is the "most stable" for me but the problem is that after a (right now it looks to me random) point in time the T265 does not deliver data anymore (poll_for_frame never returns anything after a certain while). Also tried to connect the sensor using a powered USB hub but this does not solve the problem. 
My feeling is that when I start it for the first time after boot or a longer period of not using the PI it provides data for a longer time (several minutes). When it crashes and i restart it, then it stops working after a few seconds to 1 minute. 
So could overheating be problem? I do not have active cooling with the PI?

MartyG-RealSense, 2020-02-18:
Overheating on a USB device can be caused by a fault on a USB port or a bad USB cable.  It happens fast enough that the heat should manifest within the first few minutes of using a device.  Is the RealSense device's casing hot to the touch if you touch it within a few minutes of it starting, please?

minihops123, 2020-02-18:
I know put the T265 (only the T265 without the D435i) on usb and put the raspberry pi in front of a big fan. I seems that it works "longer" -> 10 minutes. 
Without the cooling the PI gets definitly warm. CPU temp is shown close to 70 degrees. with the big fan in front of CPU temperature drops to 40 degrees.

Another hypothesis i stated now is that the T265 is standing still during the "long-term" test. And therefore it stops working. Right now it is moved together with fan :-) Lets see if this helps :D

minihops123, 2020-02-18:
Hmm.. no luck yet. So its running between 2 to 5 minutes. then it stops. Any other ideas?

MartyG-RealSense, 2020-02-18:
I am not a programming specialist.  I wonder if there is a "memory leak" going on in the script though.  A memory leak is where a program continually consumes a computer's memory capacity as it runs instead of releasing memory when it has finished a task.  This can lead to a crash of the program.  If the memory stays consumed after the crash then the program might crash sooner on subsequent runs because there is even less memory available on the computer than the first time the program was run

https://stackoverflow.com/questions/2975831/is-leaked-memory-freed-up-when-the-program-exits.

minihops123, 2020-02-18:
Hey, thanks for this idea. But I supervised the memory consumption stays constant over time. so this is not the problem.

Maybe I should focus on another type of compute unit. Maybe something like this: https://www.intel.de/content/www/de/de/products/boards-kits/nuc/kits/nuc5cpyh.html

Has anyone experience with that? The only thing that this compute unit has to do is to grab images and poses and send them via wifi

MartyG-RealSense, 2020-02-18:
NUCs are certainly very capable of being used with D435 and T265 on the same robot.  Intel have done so in a RealSense mobile robot seminar on YouTube in the link below   The part where D435 is introduced starts at 17 minutes 15 seconds into the video.

https://www.youtube.com/watch?v=62vm0_RZ1nU

Whilst there are other small compute units available, such as Intel's candy-bar sized Intel Compute Stick, I would not recommend them to be used with two RealSense devices at the same time due to power supply concerns.  A NUC or an Up Squared board (which has been shown to work with two RealSense devices) would be my primary recommendations.

minihops123, 2020-02-19:
Hey. Tried a bit around again. So I'm able at least to recover the T265. Meaning: 
poll_for_frames -> if i get a certain time no data. stop pipeline. and restart pipeline. Its not a nice way but this works for me currently. 
BTW: the issue only appears if both sensors are streaming data. If T265 is running alone, its working.

MartyG-RealSense, 2020-02-19:
Are you distinguishing between the D435 and T265 using serial numbers, please?

MartyG-RealSense, 2020-02-19:
Another user got around the problem of needing a serial number by instead programming query_devices to look for the first device with the name 'T265' in it.

https://github.com/IntelRealSense/librealsense/issues/3434#issuecomment-476290706

RealSenseSupport, 2020-10-26:
Hi @Leijoma ,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-10-30 by RealSenseSupport