realsenseai/librealsense #5510: Works from realsense viewer but throughs 'no device connected'  from python wrapper code

nilendu7980, 2019-12-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 

### Issue Description
I am unable to execute the following code. I get a 'no device found' error. Although it works from realsense-viewer. I have already tested multiple resolutions and confirmed that the device is on USB3.0

I am clueless. Can you please suggest? 

import pyrealsense2 as rs
   pipeline = rs.pipeline()
    config = rs.config()
    #config.enable_stream(rs.stream.depth,840,480,rs.format.z16,30)
    config.enable_stream(rs.stream.depth,640,480,rs.format.z16,30)
    pipeline.start(config)

Thanks,
Neel

ev-mp, 2019-12-23:
@nilendu7980 hello, please provide the missing data:
- Platform/OS/Kernel versions
- Camera type (`rs-enumerate-devices`)
- SDK and Pyrealsense versions used

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@nilendu7980 Could you please provide the information listed by ev-mp if you still need support on this? Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@nilendu7980 Could you please update? Looking forward to you reply. Thanks!

giordyb, 2020-02-10:
I'm having the same problem, realsense viewer works but python wrapper throws a "No Device connected" error.

my setup is the following:

```Name                          :     Intel RealSense D435I
    Serial Number                 :     ##################
    Firmware Version              :     05.12.02.100
    Recommended Firmware Version  :     05.12.02.100
    Physical Port                 :     2-1.3-4
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400```

Jetson Nano with Ubuntu 18.04.4 LTS
kernel 4.9.140-tegra 
python 3.6.9
SDK librealsense-2.31.0 compiled from source

ev-mp, 2020-02-10:
@giordyb , the Tegra kernel lacks IIO drivers required by the camera's IMU. In order to use it with D435i you have to compile the SDK with `-DFORCE_RSUSB_BACKEND=true` flag.

@nilendu7980, do you need support with the issue ?

giordyb, 2020-02-11:
@ev-mp thanks, that did the trick for me!

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

@nilendu7980 Could you please provide the information below?

- Platform/OS/Kernel versions
- Camera type (`rs-enumerate-devices`)
- SDK and Pyrealsense versions used

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@nilendu7980 Sorry we didn't get response from you for weeks, will close the issue accordingly. Please create another one if you still have issues or questions. Thanks!

closed 2020-03-04 by RealSenseCustomerSupport