realsenseai/librealsense #5511: Realsense viewer does not work with any setup of kernel or SDK version

FPSychotic, 2019-12-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435 t265 } | 
| Firmware Version                   | last official| 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.18 low latency or 4.18  or 5.0lts                                        | 
| Platform                                 | PC i7 8565u uhd620 gpu 8ram |
| SDK Version                            | 2.31 or 2.29                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  robot         
[image]

I installed from source or apt, 2.29 or 2.31, clean install of ubuntu, kernel 4.18, 4.18 low latency and 5.0 (last from ubuntu image), mesa drivers from apt or ppa and glxgears, glmark and glmark2 se2 works correctly. the cameras on ros wrapper works ( with connection issues when I connect 435+t265), multicam test works, rs enumerate devices works, lsusb recognice the ca,eras with ID (sometimes t265 no ID).
Realsense viewer doesnt work.
thanks by all this nice cameras and the constant support and updates, is just amazing.
              |

dorodnic, 2019-12-22:
Hi @FPSychotic 
Sorry for the issue. Please run `rm ~/.realsense-config.json` and rerun the Viewer

FPSychotic, 2019-12-22:
I did already and I did again now, I tried different resolutions too, and same issue.Thanks by the fast answser a sunday.

dorodnic, 2019-12-22:
I think F8 or F10 should switch Viewer into full screen mode, perhaps this will help?

FPSychotic, 2019-12-22:
Did not changed anything. Anyway this is not a critical thing, I think I will wait to next apt update. pleuse dont use your Sunday time on this
Thanks a lot

dorodnic, 2019-12-22:
Another, a bit hacky solution would be to add `scale_factor = 1.f;` before [imgui_impl_glfw.cpp:243](https://github.com/IntelRealSense/librealsense/blob/c3c758d18c585a237bb5b635927797aa69996391/third-party/imgui/imgui_impl_glfw.cpp#L243).
I think for whatever reason monitor physical size is not correctly identified.

FPSychotic, 2019-12-22:
Ahh, dont worry more at the moment, I have a DVI adapter on some place, the monitor is a tv via hdmi quite old, Im quite sure is due to that

RealSenseCustomerSupport, 2019-12-30:
----------------------------------------------

Hi @FPSychotic Do you need further help on this? Can we close this issue?
Thanks,

closed 2020-01-08 by RealSenseCustomerSupport