realsenseai/librealsense #5512: Support standard ROS wstool and rosdep install

tylerjw, 2019-12-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | any | 
| Firmware Version                   | any | 
| Operating System & Version |   Ubuntu 18.04 (Melodic)  | 
| Kernel Version (Linux Only)    |  5.3.0-24-generic | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C++  | 
| Segment			|  ROS  | 

### Issue Description
Currently the recommended way to install this package is through the debian package even though there is an included package.xml.  Ideally this would be a published ros package (through bloom) and in the meantime all build dependencies should be installed by rosdep install if a user puts this repo in their `rosinstall`.  By doing this, using this repo in a ROS project would be much more standard.

If I add this to my ros workspace through `wstool` and run `rosdep install` and then `catkin build` it fails with this error:
```bash
Errors << librealsense2:cmake /root/ros_ws/logs/librealsense2/build.cmake.000.log
CMake Error at /root/ros_ws/src/librealsense2/third-party/glfw/CMakeLists.txt:230 (message):
  The RandR headers were not found

cd /root/ros_ws/build/librealsense2; catkin build --get-env librealsense2 | catkin env -si  /usr/bin/cmake /root/ros_ws/src/librealsense2 --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/root/ros_ws/install -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE=-O3; cd -
```

tylerjw, 2019-12-23:
# Steps to Reproduce

```bash
docker pull ros:melodic-ros-base
docker run -it ros:melodic-ros-base /bin/bash

# Inside docker container
apt-get -qq update
apt-get -qq dist-upgrade
apt-get -qq install -y python-catkin-tools ssh git

# Craete ROS workspace
CATKIN_WS=/root/ros_ws
mkdir -p ${CATKIN_WS}/src
cd ${CATKIN_WS}/src

# checkout librealsense2 library into workspace
wstool init .
set librealsense2 --git https://github.com/IntelRealSense/librealsense.git
wstool update

# Install dependencies
rosdep install -y -q -n --from-paths . --ignore-src --rosdistro melodic

# Config and build
catkin config --extend /opt/ros/melodic --install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE="-O3"
catkin build
```

tylerjw, 2019-12-23:
I figured out what the missing dependencies are (for building on that mostly bare docker image). See my PR that fixes the package.xml file and is some de-facto documentation of dependencies.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@tylerjw Thanks for the PR contribution! We will have engineers to review the PR and feedback to you later. Thanks!

tylerjw, 2020-01-07:
@RealSenseCustomerSupport any progress on this?

tylerjw, 2020-01-11:
ping

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@tylerjw I saw our engineers have provided comments on your PR https://github.com/IntelRealSense/librealsense/issues/5512  Any other questions? Thanks!

tylerjw, 2020-02-03:
@RealSenseCustomerSupport I responded on the PR.  I still think the default build dependencies should be listed clearly.

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@tylerjw

tylerjw, 2020-03-20:
This has been fixed and we are using the released version of this now.  Closing.

closed 2020-03-20 by tylerjw