realsenseai/librealsense #5514: How to live stream Infrared frames in MATLAB?

BeatrizAMartins, 2019-12-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D430 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.22.0                         | 
| Language                            |  MATLAB     | 

### Issue Description
I am currently using the code for function depth_example

```
function depth_example()
    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    % Make Colorizer object to prettify depth output
    colorizer = realsense.colorizer();

    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();

    % Get streaming device's name
    dev = profile.get_device();
    name = dev.get_info(realsense.camera_info.name);

    % Get frames. We discard the first couple to allow
    % the camera time to settle
    for i = 1:5
        fs = pipe.wait_for_frames();
    end
    
    % Stop streaming
    pipe.stop();

    % Select depth frame
    depth = fs.get_depth_frame();
    % Colorize depth frame
    color = colorizer.colorize(depth);

    % Get actual data and convert into a format imshow can use
    % (Color data arrives as [R, G, B, R, G, B, ...] vector)
    data = color.get_data();
    img = permute(reshape(data',[3,color.get_width(),color.get_height()]),[3 2 1]);

    % Display image
    imshow(img);
    title(sprintf("Colorized depth frame from %s", name));
end

```

Is there any way to live stream Infrared frames in MATLAB?

MartyG-RealSense, 2019-12-23:
There was a past case where a RealSense user said that the code below had the MATLAB wrapper syntax that enabled them to stream unrectified Y16 in MATLAB:

****************

cfg = realsense.config();
cfg.enable_stream(realsense.stream.infrared, 1, realsense.format.y16)
cfg.enable_stream(realsense.stream.infrared, 2, realsense.format.y16)

************

cfg (config) instructions are placed before the pipe.start instruction.

JimmyFroost, 2019-12-23:
hello

Maybe you should : 
 
- remove  **pipe.stop**  , or at least put it at the end

- put the code in a while loop to have real time and live stream

Here is my config :
`config415.enable_stream(realsense.stream.depth, 1280, 720, realsense.format.z16, 30);`

Here is how I do  :

  ```
  while (app.Switch.Value =='1' )
   
        fs415 = pipe415.wait_for_frames();
        depth = fs415.get_depth_frame();
        colorDepth = colorizer.colorize(depth); 
        data415 = colorDepth.get_data();
  
       imgDepth = permute(reshape(data415',[3, dist_X , dist_Y]),[3 2 1]);
       imshow( imgDepth);
         
 end
```

Good luck : )

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@BeatrizAMartins Did you get it through with suggestions from JimmyFroost and MartyG-RealSense? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@BeatrizAMartins Any update from your side? Looking forward to your reply. Thanks!

BeatrizAMartins, 2020-02-09:
> There was a past case where a RealSense user said that the code below had the MATLAB wrapper syntax that enabled them to stream unrectified Y16 in MATLAB:
>
> cfg = realsense.config(); [...]

Thank you!! It worked

MartyG-RealSense, 2020-02-09:
@BeatrizAMartins   Great, thanks for the update!

closed 2020-02-12 by RealSenseCustomerSupport