realsenseai/librealsense #5515: Gyro Angular velocity error

mgzjzzhang, 2019-12-23:
| | |
|---|---|
|**librealsense**|2.31.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|848312071368|
|**Firmware Version**|05.12.01.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|915323050158|
|**Firmware Update Id**|915323050158|

The Gyro stream display x: y: z: axial angular velocity , but only x axial 's value is correct, y and z axial 's value is always going to be 56.802 and -22.441,no matter what.
I tried to reinstall the system and SDK, but nothing worked. Can someone help me?

ev-mp, 2019-12-23:
@mgzjzzhang  hello, 
Please generate and provide data for analysis using the supplied `rs-data-collect` [tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect):
- Create config file  `data_collect.cfg`  that contains the following two lines
ACCEL,1,1,63,MOTION_XYZ32F
GYRO,1,1,200,MOTION_XYZ32F

- Run the collection tool when the camera is idle with the command line parameters
>rs-data-collect -c ./data_collect.cfg -f ./data_log.csv -t 20

mgzjzzhang, 2019-12-24:
Thank you ,Here  is the information I collected:

Configuration:
Stream Type,Stream Name,Format,FPS,Width,Height
Accel,Accel,MOTION_XYZ32F,63
Gyro,Gyro,MOTION_XYZ32F,200

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms),3DOF_x,3DOF_y,3DOF_z

Gyro,0,1,1577145754306.990,89.362,0.002,56.737,-21.548
Gyro,0,2,1577145754307.376,89.588,0.002,56.737,-21.548
Gyro,0,3,1577145754312.185,94.388,-0.002,56.737,-21.548
Gyro,0,4,1577145754317.173,99.370,-0.002,56.737,-21.548
Gyro,0,5,1577145754322.177,104.374,-0.005,56.737,-21.548
Gyro,0,6,1577145754327.179,109.376,-0.005,56.737,-21.548
Gyro,0,7,1577145754332.143,114.361,-0.005,56.737,-21.548
Gyro,0,8,1577145754337.146,119.313,-0.009,56.737,-21.548
Gyro,0,9,1577145754342.384,124.583,-0.002,56.737,-21.548
Gyro,0,10,1577145754347.137,129.313,-0.003,56.737,-21.548
Gyro,0,11,1577145754352.429,134.635,-0.003,56.737,-21.548
Gyro,0,12,1577145754357.427,139.624,-0.007,56.737,-21.548
Gyro,0,13,1577145754362.417,144.613,-0.005,56.737,-21.548
Gyro,0,14,1577145754367.426,149.623,-0.010,56.737,-21.548
Gyro,0,15,1577145754372.426,154.633,-0.005,56.737,-21.548
Gyro,0,16,1577145754377.424,159.620,-0.002,56.737,-21.548
Gyro,0,17,1577145754382.423,164.621,-0.002,56.737,-21.548
Gyro,0,18,1577145754387.422,169.629,-0.005,56.737,-21.548
Gyro,0,19,1577145754392.422,174.618,-0.005,56.737,-21.548
Gyro,0,20,1577145754397.423,179.619,-0.009,56.737,-21.548
Gyro,0,21,1577145754402.407,184.611,-0.003,56.737,-21.548
Gyro,0,22,1577145754407.418,189.615,-0.007,56.737,-21.548
Gyro,0,23,1577145754412.420,194.620,-0.005,56.737,-21.548
Gyro,0,24,1577145754417.415,199.612,-0.003,56.737,-21.548
Gyro,0,25,1577145754422.417,204.622,-0.005,56.737,-21.548
Gyro,0,26,1577145754427.418,209.616,-0.005,56.737,-21.548
Gyro,0,27,1577145754432.416,214.613,-0.009,56.737,-21.548
Gyro,0,28,1577145754437.417,219.623,-0.005,56.737,-21.548
Gyro,0,29,1577145754442.415,224.612,-0.003,56.737,-21.548
Gyro,0,30,1577145754447.377,229.624,-0.005,56.737,-21.548
Gyro,0,31,1577145754452.417,234.624,-0.003,56.737,-21.548
Gyro,0,32,1577145754457.415,239.611,-0.005,56.737,-21.548
Gyro,0,33,1577145754462.415,244.611,-0.007,56.737,-21.548
Gyro,0,34,1577145754467.415,249.620,-0.003,56.737,-21.548
Gyro,0,35,1577145754472.415,254.611,-0.005,56.737,-21.548
Gyro,0,36,1577145754477.413,259.610,-0.005,56.737,-21.548
Gyro,0,37,1577145754482.414,264.620,-0.005,56.737,-21.548
Gyro,0,38,1577145754487.411,269.609,-0.005,56.737,-21.548
Gyro,0,39,1577145754492.410,274.606,-0.007,56.737,-21.548
Gyro,0,40,1577145754497.412,279.619,-0.005,56.737,-21.548
Gyro,0,41,1577145754502.409,284.605,-0.005,56.737,-21.548
Gyro,0,42,1577145754507.409,289.607,-0.002,56.737,-21.548
Gyro,0,43,1577145754512.411,294.620,-0.005,56.737,-21.548
Gyro,0,44,1577145754517.406,299.603,-0.003,56.737,-21.548
Gyro,0,45,1577145754522.406,304.603,-0.003,56.737,-21.548
Gyro,0,46,1577145754527.406,309.612,-0.002,56.737,-21.548
Gyro,0,47,1577145754532.406,314.603,-0.005,56.737,-21.548
Gyro,0,48,1577145754537.404,319.601,-0.005,56.737,-21.548
Gyro,0,49,1577145754542.642,324.842,-0.005,56.737,-21.548
Gyro,0,50,1577145754547.406,329.603,0.000,56.737,-21.548
Gyro,0,51,1577145754552.408,334.605,-0.009,56.737,-21.548
Gyro,0,52,1577145754557.406,339.604,-0.003,56.737,-21.548
Gyro,0,53,1577145754562.405,344.606,-0.009,56.737,-21.548
Gyro,0,54,1577145754567.404,349.600,-0.007,56.737,-21.548
Gyro,0,55,1577145754572.403,354.600,-0.005,56.737,-21.548
Gyro,0,56,1577145754577.404,359.603,-0.005,56.737,-21.548
Gyro,0,57,1577145754582.401,364.598,-0.005,56.737,-21.548
Gyro,0,58,1577145754587.401,369.597,-0.005,56.737,-21.548
Gyro,0,59,1577145754592.403,374.601,-0.002,56.737,-21.548
Gyro,0,60,1577145754597.400,379.597,-0.005,56.737,-21.548
Gyro,0,61,1577145754602.399,384.595,-0.005,56.737,-21.548
[... 3141 lines cut ...]
Gyro,0,3203,1577145770325.080,16107.278,0.127,56.737,-21.548
Gyro,0,3204,1577145770330.019,16112.277,0.126,56.737,-21.548
Gyro,0,3205,1577145770335.077,16117.276,0.318,56.737,-21.548
Gyro,0,3206,1577145770340.074,16122.285,0.480,56.737,-21.548
Gyro,0,3207,1577145770345.075,16127.274,0.679,56.737,-21.548
Gyro,0,3208,1577145770350.073,16132.277,0.925,56.737,-21.548
Gyro,0,3209,1577145770355.073,16137.271,1.143,56.737,-21.548
Gyro,0,3210,1577145770360.073,16142.270,1.257,56.737,-21.548
Gyro,0,3211,1577145770365.073,16147.271,1.265,56.737,-21.548
Gyro,0,3212,1577145770370.072,16152.270,1.241,56.737,-21.548
Gyro,0,3213,1577145770375.070,16157.267,1.093,56.737,-21.548
Gyro,0,3214,1577145770380.073,16162.271,0.937,56.737,-21.548
Gyro,0,3215,1577145770385.072,16167.270,0.803,56.737,-21.548
Gyro,0,3216,1577145770390.066,16172.263,0.667,56.737,-21.548
Gyro,0,3217,1577145770395.068,16177.267,0.543,56.737,-21.548
Gyro,0,3218,1577145770400.026,16182.224,0.466,56.737,-21.548
Gyro,0,3219,1577145770405.068,16187.267,0.387,56.737,-21.548
Gyro,0,3220,1577145770410.064,16192.263,0.340,56.737,-21.548
Gyro,0,3221,1577145770415.070,16197.269,0.360,56.737,-21.548
Gyro,0,3222,1577145770420.072,16202.270,0.380,56.737,-21.548
Gyro,0,3223,1577145770425.064,16207.263,0.435,56.737,-21.548
Gyro,0,3224,1577145770430.064,16212.262,0.541,56.737,-21.548
Gyro,0,3225,1577145770435.064,16217.262,0.726,56.737,-21.548
Gyro,0,3226,1577145770440.064,16222.264,1.007,56.737,-21.548
Gyr

ev-mp, 2019-12-24:
@mgzjzzhang thanks you for the info.
I can confirm that this is and SDK issue introduced in the latest version. This is is a high-priority task and we will address it in the next release.In the meanwhile please continue using v2.30.0.

Sorry for the inconvenience.

ev-mp, 2020-01-26:
@mgzjzzhang hello, this should be fixed in v2.32.1.
Please check and update.

mgzjzzhang, 2020-01-26:
Ok, thanks. I'll try

closed 2020-02-04 by ev-mp