realsenseai/librealsense #5516: Unity asset does not work on android

MubasshirAhmed, 2019-12-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MubasshirAhmed, 2019-12-23:
I have build a apk and install it on an android phone. But it does not working. I have build it using unity asset.

crogersarvr, 2019-12-24:
Me, too.  I built the android native/java examples, they work on android10/pixel3xl and oculus quest! (custom android 7.1.1).  The app displays the camera feed and depth feed in a VR window on the quest.

I built the librealsense android module .aar file.  I built android apps with unity2019.3 using the .aar file.  Everything compiled and executed on the quest/pixel3 but the unity app would not detect the camera.  ADB Logcat lists the camera and its connection events correctly, i dont see an obvious Unity error in Logcat.  
When i built just the pointcloud scene, the app would alternate between a black screen with Oculus "wait dots" (device not plugged in ) and a blue screen with no text (its really a skydome on the quest rather than a screen) when plugged in.  Building all the scenes, the starter scene would just say "connect a device".

realsense 435i
firmware: 05.11.15.00 (i should try newer firmware...)
new usb 3.0 cable 
Windows10 (build)  android7.1.1/quest and android10/pixel3xl (deploy)
NDK 20 something (im pretty sure unity HAS to use ndk 16b, i should try building the non-unity android app with ndk16...)
librealsense from github a few days ago

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@MubasshirAhmed  @crogersarvr Currently there's no sample code for Unity connected to librealsense on android, but you can call Java API in Unity code to do some testing. Also please check USB permission.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@MubasshirAhmed  @crogersarvr  Any other questions about this ticket?Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@MubasshirAhmed  @crogersarvr  Sorry that we didn't get response from you for weeks, will close this issue at this point. Please feel free to create new one if you still have issues or questions.Thanks!

closed 2020-03-04 by RealSenseCustomerSupport