realsenseai/librealsense #5518: Could not find file "config.h"

YuzheHao, 2019-12-23:
I downloaded the master-branch, and try to run a depth example program in Clion, then I got this error message:
> librealsense-master/cmake-build-debug/third-party/libusb/libusb/core.c:23:20: fatal error: config.h: No such file or directory #include "config.h"
Before I downloading the whole master-branch, I have tried to run a program to get the distance, and it works, so I don't think my environment is not complete.
Maybe it is a rather beginner problem but I hope could get some helps here

dorodnic, 2019-12-23:
Hi @YuzheHao 
Have you tried manually running `cmake .. && make`? I suspect something is wrong with Clion CMake configuration, you should not be encountering this error when building from source.

YuzheHao, 2019-12-24:
Thank you for the answering,
i tried to run `cmake .. && make`, and I got the following error:

> CMake Error at CMakeLists.txt:5 (include):
>  include could not find load file:
>     CMake/lrs_options.cmake
> 
> CMake Error at CMakeLists.txt:6 (include):
>   include could not find load file:
>     CMake/connectivity_check.cmake
> 
> CMake Error at CMakeLists.txt:9 (if):
>   if given arguments:
>     "OR" "OR"
> 
>   Unknown arguments specified
> -- Configuring incomplete, errors occurred!
> See also "/work/yuzhe/docs/librealsense-master/CMakeFiles/CMakeOutput.log".

Do you have any idea how could I check the cmake config of CLion?

dorodnic, 2019-12-24:
Something is strange with the output, step by step, please try:
```
cd ~
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build
cd build
cmake ..
make
```
This should download and build librealsense on your machine

Esquimen, 2019-12-30:
Install this lib for debug:
sudo apt-get install libusb-1.0-0-dev

YuzheHao, 2020-01-06:
> Something is strange with the output, step by step, please try:
> 
> ```
> cd ~
> git clone https://github.com/IntelRealSense/librealsense.git
> cd librealsense
> mkdir build
> cd build
> cmake ..
> make
> ```
> 
> This should download and build librealsense on your machine

Sorry.... I tried this line by line, and it is still telling me "lacking config.h". And this time I ran all these commands in shell rather than in CLion, which makes me confused more....

YuzheHao, 2020-01-06:
Hi everyone, I have solved this problem, thank you for your helping.
Later I will summarize the solution and post it here.

Jiloc, 2020-01-13:
To me, running 

> sudo apt-get install libusb-1.0-0-dev

solved the issue

closed 2020-01-23 by RealSenseCustomerSupport

ernestp, 2020-10-28:
I'm getting this error during cross compile. 
Is there any workaround to compile libusb without installing globally?

monacv, 2021-01-14:
`$  sudo apt-get install libusb-1.0-0-dev`
didn't solve the issue for me 
https://pastebin.com/raw/aNeb2KLB

ddwell, 2021-03-11:
@monacv try cleaning the build folder and doing cmake && make again
`sudo rm -R build
mkdir build && cd build
cmake ..
make
`